LumaTeam/Luma3DS #7: Aureinand booting problems.

fallenlucario, 2016-03-29:
Hello! I'm not sure if you read my comment on the Aureinand thread on gbatemp

I've been using this guide to manually compile my own A9LH.3dsx files.
https://gbatemp.net/threads/how-to-compile-and-install-arm9loaderhax.415776/

So my problem here is that every version of Aureinand after 3.11 does not load on my A9LH setup. I used FIX94's fork of A9LH to compile my .3dsx file. The only reason I used his is to not have the backlight flicker on then off during screen init. I assume I need screen init in order to boot various other programs like D9 and other things.

I think that this may have something to do with the merging of the loader file and other things in the aurei folder into the main executable as I'm only having problems with versions that don't have the files in the aurei folder.

Not sure if you'd be willing to fix this so it works on my version of A9LH or if you'd recommend I change my setup. There are so many forks of A9LH that I honestly don't know what to do or what each of them does differently. If you can recommend me a setup that has screen init with no backlight and works with your latest releases of AuReiNand that'd be amazing! Thank you for all of your hard work!

fallenlucario, 2016-03-29:
It seems the backlight needs to be enabled for the inital boot of the newer aureinand's to configure the menu for it. Afterwards it runs fine without it!  Sorry to bother you but would you mind informing me what the prepatched firms on this menu and the firm9.0 mean? 

For some reason when i used firm9.0 it boot me into a blank NAND, it might have been my sysnand because my emunand has a theme on it.

astronautlevel2, 2016-03-30:
@fallenlucario 
The prepatched firms option means that after AuReiNand applies its patches to the firmware, it saves it on your SD card rather than patching it every time. This marginally increases boot speed, at the expense of a few megabytes on your SD card.

Using firm9.0 means that you use the NATIVE_FIRM from 9.0. This was the last NATIVE_FIRM where we had arm9 kernel exploits, which lets us launch CFW, boot Gateway, run things like Decrypt9, etc. This NATIVE_FIRM, however, doesn't work with anything newer than 9.3, so if you're booting the firm9.0 it will launch your (probably 9.2 from what it sounds like) SysNAND.

ApacheThunder, 2016-03-31:
I can confirm this as an issue too. It will hang and fail to init screens if used with Fix94's version of A9L where the backlights are off by default. I got around this by using Arm9Select to enable backlights so the config menu will come up.

Also an issue. If using Fix94's A9L, backlights remain off even if a splash file is specified, so I still have to use Arm9Select to enable the backlights. At this point I have no plans on changing my internal A9L payload again. So unless this issue is fixed I guess I'm stuck booting it with Arm9Select.

piratesephiroth, 2016-03-31:
works perfectly fine with Aurora's A9LH fork

AuroraWright, 2016-04-02:
Doing screen init but leaving backlight at 0x0 is a non-standard environment. The proper way for Fix94 would be to NOT do screen init in stage2 and doing it fully from inside Arm9Select. This way detection wouldn't be borked. This is not an issue with Aurei itself.

closed 2016-04-02 by AuroraWright

ApacheThunder, 2016-04-02:
"funny enough that way around arm9select always hung for me and I didnt understand at all why. "

Quote from Fix94. Guess this will be a problem everyone will have to live with. :(