LumaTeam/Luma3DS #10: Building: makerom: -nocodepading not found

mariogamer2, 2016-04-02:
Latest makerom...

AuroraWright, 2016-04-02:
You need it to build it from source. That's why I said "a recent build".

closed 2016-04-02 by AuroraWright

mariogamer2, 2016-04-02:
build it from source? from 3ds_injector?

AuroraWright, 2016-04-02:
Then it works. I meant makerom.

mariogamer2, 2016-04-03:
?

mariogamer2, 2016-04-03:
makerom -f ncch -rsf ../loader.rsf -nocodepadding -o /home/**/AuReiNand/injector/injector.cxi -elf /home/***/AuReiNand/injector/injector.elf
[SETTING ERROR] Unrecognised argument '-nocodepadding'
[RESULT] Invalid arguments, see 'makerom -help'
/home/***/AuReiNand/injector/Makefile:121: recipe for target '/home/***/AuReiNand/injector/injector.cxi' failed
make[2]: *** [/home/***/AuReiNand/injector/injector.cxi] Error 252
Makefile:103: recipe for target 'build' failed
make[1]: *** [build] Error 2
make[1]: Leaving directory '/home/****/AuReiNand/injector'
Makefile:74: recipe for target 'build/patches.h' failed
make: *** [build/patches.h] Error 2

mariogamer2, 2016-04-03:
It's strange...

When doing the command by myself,it's work but not the makefile...

JixunMoe, 2016-04-13:
I made a few lines of modification and build with Msys's gcc:

Repo: [profi200/Project_CTR](https://github.com/profi200/Project_CTR/tree/master/makerom)

```
--- a/makerom/utils.c
+++ b/makerom/utils.c
@@ -185,7 +185,7 @@ bool AssertFile(char *filename)
 {
        if(filename == NULL)
                return false;
-#ifdef _WIN32
+#ifdef _WIN32_x
        struct _stat64 st;
        return _stat64(filename, &st) == 0;
 #else
@@ -196,7 +196,7 @@ bool AssertFile(char *filename)

 u64 GetFileSize64(char *filename)
 {
-#ifdef _WIN32
+#ifdef _WIN32_x
        struct _stat64 st;
        if( _stat64(filename, &st) != 0)
                return 0;

--- a/makerom/oschar.h
+++ b/makerom/oschar.h
@@ -36,8 +36,8 @@ typedef char oschar_t; // UTF8
 #define os_closedir _wclosedir
 #define os_chdir _wchdir

-#define _osstat _stat64
-#define os_stat _wstat64
+#define _osstat stat
+#define os_stat stat

 #define os_fopen _wfopen
 #define OS_MODE_READ L"rb"
```