LumaTeam/Luma3DS #11: Updating titles not working on >=4.0

opa334, 2016-04-02:
If you try and update a title (Flipnote Studio 3D in my case) from system menu, it will tell you that your firmware is outdated and it will try to update and find nothing. I have downgraded to 3.14 and performed the update without issues. I think this might be related to some 3ds_injector stuff added.
New 3DS (EUR) 10.7 SysNand using A9LH

AuroraWright, 2016-04-02:
Interesting:
1) eShop works fine?
2) Can you try one of these builds? http://www51.zippyshare.com/v/G9bZaSLN/file.html
http://www51.zippyshare.com/v/P0Ve4NWs/file.html
I'm pretty sure one of those fixes the issue.

AuroraWright, 2016-04-02:
Can't reproduce. I updated to 10.7, got into eShop to enable home menu updates, rebooted, and updated from Home Menu 3 cartridge games. No error whatsoever. Latest code from github (nothing should have changed since 4.1).

AuroraWright, 2016-04-02:
I reproduced it, but it hasn't to do with updates. It has to do with full titles downloads from Home Menu.

theAeon, 2016-04-06:
Just to further confirm, I attempted to press download now on a full system title and it prompted me to update on aurei 4.2

Shadow322, 2016-04-07:
This has already been stated as a issue to sole it click open eshop for
update

giveup, 2016-04-13:
Are you enable "user developer unitinfo"?
This option will cause any Nintendo service error.(eShop/Miiverse)

theAeon, 2016-04-13:
Nope, I have not.

-Andrew

cspencer49519, 2016-04-15:
I have one game that will not update, but I suspect maybe a bad CIA, I get a problem where the update stops at 98%. The title is Marvel Pinball, and I believe it is E-Shop only.

RShadowhand, 2016-04-16:
I need diagnostic information, so everyone who has this problem, please answer these questions:

A9LH?
Patched TWL/AGB Firms? (not by CFW, but by installing CIA)
Patched Home Menu/NS? (by installing patched CIAs, for region-free)
Do you own the CIA? (don't post links)
Did you use PlaiCDN to download the CIA? (don't post titlekeys)

cspencer49519, 2016-04-16:
I have an N3DSXL, A9LH, no patched TWL/AGB, no patched Home Menu/NS, the cia came from "that iso site", I do not own it, but I can buy it if it will help with diagnostics. I did not use plaicdn to download.

theAeon, 2016-04-16:
Aurei 4.2, A9LH, no patched cias, title was pushed from my Nintendo.

-Andrew

cspencer49519, 2016-04-16:
If you need me to do anything diagnostic, including buying the game and testing downloading/updating after that, just let me know, I'm available.

RShadowhand, 2016-04-16:
@theAeon What do you mean "pushed from My Nintendo", you mean it's legit and you got it from My Nintendo?

theAeon, 2016-04-16:
I used the send game to device feature off their website. I believe that it tells the device to use download later on the title, and pressing download now from the home menu prompts for a sys update.

RShadowhand, 2016-04-16:
@AuroraWright I think it has something to do with firm patching on the fly. I thought I had this issue because I had prepatched TWL/AGB FIRMs and prepatched HomeMenu / NS.

cspencer49519, 2016-04-16:
It may be worth noting that I had the same issue with this title when AuReiNAND used to create it's own pre-patched firms in the aurei directory, so it might not have anything to do with patching on the fly, but maybe just the way it's being patched in general? I'm pretty clueless but just thought I would throw that out there.

theAeon, 2016-04-17:
Just updated to 5.1 on vanilla a9lh, still doesn't work with titles that have the download started from Nintendo.com- twilight princess picross in this case

theAeon, 2016-04-17:
^vanilla aurei a9lh that is

RussKay, 2016-04-20:
Same here. Pushed from Nintendo to homescreen. On A9LH (latest noscreeninit) and latest ARN. Selecting to download now prompted me to upgrade my system. However, I was able to get it to download by putting my system in sleep mode.

RShadowhand, 2016-04-20:
I found the issue is caused by eShop check patch, and removing it fixed it. I made a PR, so you'll have to wait a bit.

hartmannaf, 2016-04-20:
AuroraWright already knows about this. Till there is a better patch, simply update full update titles(like Flipnote studio)  from the eshop instead from the home menu.

cspencer49519, 2016-04-20:
The game I'm trying to update, Marvel Pinball, will not update through any method. It will not download through the home menu, eshop, or while in sleep mode.

cspencer49519, 2016-04-20:
Also I don't think that PR is going to be accepted, since you rebranded the entire firmware to your own name (ShadowFW?) lol.

RShadowhand, 2016-04-20:
I made the PR before I renamed it. Doesn't really matter, because it's only like 10~ lines.

hartmannaf, 2016-04-20:
@cspencer49519 does it work without the update patch? Do you have it originally from the eshop, or have you installed it using a cia?
If you installed it via cia it will not update, unless you have a legit ticket.

cspencer49519, 2016-04-20:
Installed via cia, as I stated in my previous posts. Everything else I have installed from cia updates fine. I'm not sure what version doesn't have the update patch, so unsure how to test that.

RShadowhand, 2016-04-20:
The problem arises from the "eshop update check" patch. Once disabled, it works fine.

cspencer49519, 2016-04-20:
Please tell me where I can disable this patch?

RShadowhand, 2016-04-20:
You can't without a special build. I'll build you one until there's a new release.

cspencer49519, 2016-04-20:
Awesome, I just have the one title that won't update, it's annoying but causes no problems beyond the annoyance.

hartmannaf, 2016-04-20:
Could it be possible that the other games where games like pokemon that have an  extra update title and not directly update the game files? Because since its not even updating from the eshop I  don't think its the patch.

RShadowhand, 2016-04-20:
[AuReiNand.zip](https://github.com/AuroraWright/AuReiNand/files/228454/AuReiNand.zip)

No need to rename/remove anything, just replace the arm9loaderhax.bin file.

cspencer49519, 2016-04-20:
Game still will not update, tried sleep mode, update from home, and update from eshop, I'm just going to chalk this up to a bad cia? I mean it's really not that significant, it's just something I noticed, if I ever decide to play the game I'll just buy it, I'm uninstalling it for now.

RShadowhand, 2016-04-20:
Yeah it's 100% bad cia. I know that patch works because I used literally the same build to test getting MegaMan from eShop (I own it), and said sleep mode, then simply went to home menu and said "download now", and it downloaded without issues.

RShadowhand, 2016-04-20:
Made a proper pull request (#46)

RShadowhand, 2016-04-26:
Fixed in commit d40722af5367c0410477cd7ad26a1dffb0c7d100. Wait for the new release.

closed 2016-04-26 by RShadowhand