LumaTeam/Luma3DS #12: Region Free Loading Issue?

fallenlucario, 2016-04-04:
So I'm not sure why this is happening but I'm using aureinand 4.1 and I've installed an animal crossing new leaf cia (JPN) I can see it on the home menu and it looks fine, the banner doesnt act up at all. When i try to load it it loads and then it boots me to the home menu and says that an error has occurred and that it will reboot. It's weird because loading a japanese game from a cart works fine. I'm not sure what's wrong.

nobitagamer, 2016-04-06:
Yeah, the 4.2 does not inject SecureInfo_C (in CTRNAND/rw/sys/SecureInfo_C) correctly like 4.1 so i revert back to 4.1

nobitagamer, 2016-04-07:
I found the solution, just revert file injector/source/patcher.c to revision [956829864cc90e4189eb66a0e0fd3998d75eef53](https://raw.githubusercontent.com/AuroraWright/AuReiNand/956829864cc90e4189eb66a0e0fd3998d75eef53/injector/source/patcher.c) everything will be fine.

AuroraWright, 2016-04-07:
SecureInfo_C has nothing to do with region free of games. And you can copy it easily with GodMode9.

closed 2016-04-07 by AuroraWright

fallenlucario, 2016-04-13:
Where could i find a compatibility list for the Freemultipatcher?