LumaTeam/Luma3DS #15: Sometimes themes are lost on reboot.

WobbleTheHutt, 2016-04-06:
Since 4.01 I have had issues with my themes being lost on reboot sometimes. I have yet to delete my theme management data and rebuild it. I will try this and update

AuroraWright, 2016-04-07:
I can't reproduce it, and no patches have to do with themes (the home menu is even only patched for RF). Make sure you don't use eShop or an eShop applet or that will remove your DLC and themes.

closed 2016-04-07 by AuroraWright

WobbleTheHutt, 2016-04-07:
All the themes I have are purchased from the eshop legitimately. I've also had basic themes get turned back to white. I noticed it first after restarting from a ds game and or dsi ware. (running updated sysnand with a9lh). I'll blow out my theme database today and see if that fixes it and also try a rollback as if it persists it could be related to using a 200gb micro SD card.  Thanks for your time and effort!

albertosramires, 2016-04-07:
I also have this issue.

WobbleTheHutt, 2016-04-08:
Verified card was good, reformatted with 64KB sectors rolled back to the
last release of the 3.xx branch. Going to perform 25 power cycles, 25 ds
game launch and quit, 25 gba launch and quit. (manually patched gba firm)
and 25 dsi ware launch and quit. I'll then move to 4.2 and repeat and
report findings.