LumaTeam/Luma3DS #17: Freezes pre-backlight init when SD card is write locked

ksanislo, 2016-04-08:
If your SD card is write locked via the slide switch, AuReiNand doesn't present any kind of notice when trying (and failing) to boot an emuNAND. It would be nice if AuReiNand could detect this as an error condition and notify the user instead.

TuxSH, 2016-04-08:
```c
bool isSDCardWriteProtected = (*(vu16*)(SDMMC_BASE + REG_SDSTATUS0) & TMIO_STAT0_WRPROTECT) == 0
```
for the detection part

ksanislo, 2016-08-26:
FWIW, this is a non-issue after switching to a newer build of A9LH that can boot without the SD card.

For anyone else who's looking for a fix for this, I'm using [this build](https://www.reddit.com/r/3dshacks/comments/4w1ysv/the_last_time_i_was_here_a9lh_had_just_come_out/d63mt11) of Aurora's A9LH.

TuxSH, 2016-08-30:
Also, the SD write-protect is only enforced by software, not by hardware. I haven't found the patch for it though.

Hikari-chin, 2017-07-08:
A warning message about this will be implemented. Thank you for reporting.

closed 2017-07-08 by Hikari-chin