LumaTeam/Luma3DS #22: Problems with Software Management

tranaconda, 2016-04-12:
When I go into Settings > Data Management > 3DS > Software, it starts enumerating but stops in the shortly after, showing "Could not access microSD card." next to the microSD card icon on the top screen.

This has been a bug for a while in open source CFWs - the same EmuNAND (back then, now I only use SysNAND) in Gateway Mode did not have this bug. No that I switched to A9LH, I can't easily switch to GW to access Software Management.

I don't think it's the SD card, as I tested it thoroughly with badblocks and used different ones as well. Also, it does not happen on my 2 O3DS.

The NAND was transfered from an O3DS and downgraded once. As stated above, in Gateway Mode before the switch to A9LH this bug did not appear, but at the same time when in rxTools.

Is it possible to make an option to disable all patches and just boot the homemenu clean?  Maybe this way we can find out what's wrong.

jatikeren, 2016-04-12:
i dont have issue with this one...

mine using a9lh with aureinand 4.2

tranaconda, 2016-04-12:
Like I said, my 2 O3DS don't have this bug either and it doesn't happen in Gateway Mode. So Gateway does something different (maybe better, maybe worse) than rxTools and AuReiNAND.

hartmannaf, 2016-04-12:
Maybe it an application with a broken header, while experimenting with modifiing the sd cards data I had a similar problem on sysnand, but using gateway it worked.

tranaconda, 2016-04-12:
Could be, but how do I find out which title is the culprit? And how does Gateway do it?

hartmannaf, 2016-04-12:
Depending on how much titles you have installed, the easiest way could be, to simply backup the nintendo 3ds folder and delete some titles using fbi, and recheck software management.

tranaconda, 2016-04-12:
A lot! I've used this NAND since I bought my first 3DS when it came out. I just noticed that HANS seems to have a hiccup now, too. I know there's probably a corrupt title to blame, but it would be nice if AuReiNAND would handle it like Gateway, right?

tranaconda, 2016-04-12:
For the record, I installed the microSD Manager as a cia a long time ago (https://gbatemp.net/threads/microsd-management-on-homecreen-and-working-on-old3dss-too.383892/page-3).

Title ID: CTR-N-HAJP

Just delete it with FBI or whatever. But it would still be nice to have a graceful fail like Gateway has.

RShadowhand, 2016-04-19:
@tranaconda Is this issue fixed?

closed 2016-04-19 by RShadowhand

tranaconda, 2016-04-19:
Well, for me personally, yes. But for the long run, it might be nice to handle faulty title gracefully like Gateway does.