LumaTeam/Luma3DS #23: A9LH Netloader Companion support?

d0k3, 2016-04-15:
You migth have already seen this:
https://github.com/d0k3/A9LH-Netload-Companion

Now, could you add support for it?

I suggest it work like this:
I change A9LH Netloader Companion to write the payload to your payload directory as nc.bin (or anything else of your choice). Upon starting, you let AuReiNand check if that file (nc.bin) exists. If it does, rename it to nc_old.bin, then start it. The rename is so that upon the next start, the default behaviour is restored again and the same testing payload won't be loaded a second time.

This support will be very helpful for ARM9 developers, as it would allow us to test payloads in the same way as HBL + 3DSlink allowed us in the past.

TuxSH, 2016-04-15:
As a temporary solution, you could maybe write the downloaded file to ```/aurei/payloads/a.bin``` or ```b.bin```

d0k3, 2016-04-22:
Adding to this, I could also make A9LH Netloader Companion (ugh, I need to change that name...) set some flag in memory, which would be kept upon reboot. ARN would only have to check for that flag and try to load the paylaod if it is set, otherwise not. That way, 'normal' users would not be affected in any way.

TuxSH, 2016-04-22:
This is pretty much what I'm doing in fdec433c67643ee78a52f2d04aa4a492b2045472

AuroraWright, 2016-04-26:
@d0k3 The memory value idea seems good. Can you write to an arbitrary address from ARM11 userland?

d0k3, 2016-04-26:
It actually depends... FCRAM should be no problem at all (don't know if that survives the reboot), other areas I'm unsure about.

AuroraWright, 2016-04-27:
@d0k3 I think the difficulty is finding a fcram spot which doesn't have FIRM/system modules code running in it, if you overwrote stuff (even if for a short time before a reboot) that would be dangerous I think.

d0k3, 2016-04-28:
With the dev version now out, you could also enable checking for that nc.bin payload (I know, this will slow down things).

I already do it in a different way in A9NC now, which might be a valid alternative: A9NC waits for user input after it received the payload, <B> to quit, any other to reboot. Now, I also write the payload to luma/payloads/left.bin, and if I press left and hold it, it reboots into the payload just fine.

Only two problems with that I see right now is [1] there is a delay in the reboot function (can't do anything about this), which means the button must be held for several seconds and [2] devs might want to use the 'left.bin' slot for something else.

Margen67, 2017-05-20:
A9LH support has been removed.

closed 2017-05-23 by TuxSH