LumaTeam/Luma3DS #25: (Enhancement) Per Game n3DS CPU Clock

cheatfreak47, 2016-04-15:
I propose it be possible to configure what games use the n3DS Clock Speed, done via a text document, similar to how you're handling region/language emulation. 

TitleID's followed by a number for what setting to use. 1 for Clock, 2 for L2, 3 for Clock+L2
Something like that.

If no Title ID is listed then the game will follow it's default behaviour.

The n3DS Clock Speed is a great feature but seamless integration would be a lot better then "always on or always off" I feel.

I haven't a clue how possible this is, but I figured I'd bring it up anyway, I really think it'd be a worthwhile addition.

TuxSH, 2016-04-15:
Not doable since we currently patch NS ("Nintendo Shell", the system module responsible for launching processes) is patched at boot.

closed 2016-04-15 by AuroraWright

cheatfreak47, 2016-04-23:
I mentioned this on the IRC, while I know it's not doable with the current patch
Is it possible if the existing method is scrapped in favor of a new method? (I know that's a long shot but bear with me)

A potentially good thing to look into with regard to that is Apache Thunder's n3DS Unleashed Patches, which you install as modded updates.

I think it'd be a worthwhile feature if it could be done efficiently.

The n3DS Clock Speed setting is nice, but having to enter the AuReiNAND config every time isn't exactly intuitive.

I don't need the 3x Clock Speed in say, Picross E, but I do in say, Sonic Lost World or Majora's Mask 3D, per game settings for this option would be the most seamless way to implement this feature and I feel that seamless, in concept is really in line with the goals of Luma3DS as a whole.