LumaTeam/Luma3DS #26: System Crashes on ARN 4.0+ at Boot

Melon-Bread, 2016-04-15:
- Console = Blue JPN N3DS LL
- sysNAND = 9.2 JPN
- emuNAND = 10.7 JPN (Made with emuNAND9)
- loader = a9lh

Ever since ARN 4.0 and above (nightlies included) my N3DS crashes after a second or two after booting my homescreen. It seems to load up fine showing my theme like normal, then proceeds to crash to this screen shown [here](http://imgur.com/GOQ5qKi). I get this no matter what NAND I boot into regardless if it is sysNAND or emuNAND. I have all of the right files in the right places (I know this because ARN 3.14 works just fine), and my a9lh work fine because I can boot other payloads with no issue (BootCTR9, D9, etc.).   
  
What makes this issue even more odd for me is that is displays the error message in English. If I were to trigger the same kind of error screen by different means (like trying to run a poorly made .CIA) I would get the same error in Japanese. I can not even begin to wonder why is hows the message in English since i have not changed or installed any CIAs to either of my NANDs.

I am by no means an expert but have come to the conclusion it has to do something with the _3ds_injector_ code that was added and used on ARN 4.0+ not liking something on my system for some reason. I say this because when loading a [3ds_injector .cakes based on ARN ](https://github.com/dankzegriefer/3ds_injector/releases/tag/1.0)implementation of it in CakesI experience the same issue. I am not sure where the issue lie in the 3ds_injector code in ARN because if I use the [3ds_injector.cakes mid-kid released ](https://gbatemp.net/threads/release-cakesfw.391200/page-139#post-6212124)a few days prior I don't have the crash at system start up, but the region free works just fine.

I have even tried coming through my installed titles on my NAND and they all match based on what 3DSbrew said I should have. I have also gone through looking for any bad installed CIAs on my SD and even uninstalled some that I thought may trigger it. Yet, I am left with the same problem.

Thank you for your time.

RShadowhand, 2016-04-16:
Have you tried renaming/moving the "Nintendo 3DS" folder in your SD?

Wolfvak, 2016-04-16:
Try deleting the /aurei/config file, I've replicated something similar by using an (intentionally) corrupted config file

Melon-Bread, 2016-04-16:
> Have you tried renaming/moving the "Nintendo 3DS" folder in your SD?

No, wouldn't that mess up my titles?

RShadowhand, 2016-04-16:
It will mess with your titles, but it's for debugging.
Just rename "Nintendo 3DS" folder to "Nintendo 3DS_", and try booting.

Melon-Bread, 2016-04-16:
Okay So i renamed it to "Nintendo 3DS_" as you said. I booted up the system like normal. It booted to the default white theme (as expected), but still after 2 seconds it crashes to the screen posted earlier.

RShadowhand, 2016-04-16:
If you're using ARN5, can you enable "Use developer UNITINFO", and then try again?

Melon-Bread, 2016-04-16:
> If you're using ARN5, can you enable "Use developer UNITINFO", and then try again?
Alright so I made sure that i was using ARN 5, went into my settings and enabled "Use developer UNITINFO" (if it matters the only other setting I have enabled is force A9Hl detection & show the system version in system settings). I then booted the console to get the same crash. [Here](http://imgur.com/0E5V49I) is a picture with the extra info at the top

RShadowhand, 2016-04-16:
Is there any way you can check the version of 0004013000003802? FBI 2.0.x is capable of doing that.

Melon-Bread, 2016-04-16:
> Is there any way you can check the version of 0004013000003802? FBI 2.0.x is capable of doing that.

Using FBI 2.0.4, [this](http://i.imgur.com/HXH8ocM.png) is the information on 0004013000003802.

RShadowhand, 2016-04-16:
<Steveice10> that error is from failing to get the serial number from the config service, i think
<Shadowhand> Steveice10, so NAND is fucked?
<Steveice10> or maybe not necessarily
<Steveice10> there are multiple places it can come from it seems
<Steveice10> getting serial no, getting region, etc
<Steveice10> basically its a general "could not get info from config service" error

Melon-Bread, 2016-04-16:
> basically its a general "could not get info from config service" error

So  what is the (or is there) solution?

RShadowhand, 2016-04-16:
Try [this](https://3ds.secretalgorithm.com/out.zip) build, see if that changes anything.

Melon-Bread, 2016-04-16:
> Try this build, see if that changes anything.

Yup the build works as it should. Everything loads up, region free is fine, & no errors.

RShadowhand, 2016-04-16:
Well, keep using that version until this is fixed. I'll inform AuroraWright.

Melon-Bread, 2016-04-16:
> Well, keep using that version until this is fixed. I'll inform AuroraWright.

Alright thank you very much for your time, and providing me a build that works on my system

AuroraWright, 2016-04-16:
If you could send me your secureinfo_a (privately, don't post it in public as it contains unique infos) I could test that. You can use GodMode9 to dump it. It's in CTRNAND/sys/rw.

Melon-Bread, 2016-04-16:
> If you could send me your SecureInfo_A (privately, don't post it in public as it contains unique infos) I could test that. You can use GodMode9 to dump it. It's in CTRNAND/sys/rw.
I PMed you on gbatemp with what you requested.
I pulled it from my emuNAND using the latest GodMode9 as you suggested.

AuroraWright, 2016-04-17:
OK, this is all very weird and I can't replicate it. I have a Japanese 10.7 EmuNAND on my N3DS, I just replaced my own secureinfo_a with yours, and it **works perfectly fine** (so the secureinfo_a signature patch isn't at fault here).
Stupid question: by any chance, do you also have a SecureInfo_C where the SecureInfo_A is? Also, did you happen to install a pre-patched region-free Home Menu/NS or something like that, before ARN had region free built-in?
I have a feeling something is wrong with the original NAND you also cloned EmuNAND from, but I can't verify that from here lol.

Melon-Bread, 2016-04-17:
> Stupid question: by any chance, do you also have a SecureInfo_C where the SecureInfo_A is?

So I just checked via latest GodMode9 at _EMUNAND CTRNAND > rw/sys/ _and it seems there is a file named "Secureinfo_C" below "Secureinfo_A".

>Also, did you happen to install a pre-patched region-free Home Menu/NS or something like that, before ARN had region free built-in?

I mean not that I know of, I never on purpose installed anything like that. For the longest time I just used gateway so RF stuff was never needed to run anything to play non JPN/RF titles.

> I have a feeling something is wrong with the original NAND you also cloned EmuNAND from, but I can't verify that from here lol.

Well i appreciate the help. Is there any way to make fresh NANDs and then from there re inject my (I may be using the wrong word here) titles so they show up and such?

RShadowhand, 2016-04-17:
@AuroraWright Steveice10 said the problem lies within SecureInfo_C.

piratesephiroth, 2016-04-17:
Uh, I've seen people talking about this SecureInfo_C file but never any explanation of its purpose (no info on 3dbrew about it and I don't have it on my 3DS)
What's it? Simply a copy of SecureInfo_A?
Why does it exist?

AuroraWright, 2016-04-17:
Delete the SecureInfo_C file. It was created because you used the original yifan injector, and now AuRei keeps loading it. I've disabled this feature in all builds since 4.1. SecureInfo_C is a copy, only useful for region changes.

closed 2016-04-17 by AuroraWright

RShadowhand, 2016-04-18:
@Melon-Bread Did this work?

Melon-Bread, 2016-04-18:
Yes I removed SecureInfo_C from my sysNAND & emuNAND and everything worked as it should. No need for a custom build.

RShadowhand, 2016-04-18:
Perfect. Happy gaming.