LumaTeam/Luma3DS #27: DS cartridge & flashcart won't boot using ARN v5

thaikhoa, 2016-04-16:
Black screens

cheatfreak47, 2016-04-16:
I use v5 and have no such issue, probably a user specific issue.

hartmannaf, 2016-04-16:
For me twl is working using the arm9loaderhax.bin, but its not working using the AuReinand.dat(n3ds, bootctr9, a9lh)

thaikhoa, 2016-04-16:
Solved by changing path for AuReiNand.dat using command line "aurei/"

AuroraWright, 2016-04-16:
AuReiNand.dat is only supported for *hax users. I say so in the wiki too. If you use A9LH the arm9loaderhax.bin will always take precedence. So not really a bug...

closed 2016-04-16 by AuroraWright

thaikhoa, 2016-04-16:
I thing is I really don't want to copy, paste & rename many arm9loaderhax.bin(s) or rename for payloads around when updating things... that's why I prefer AuReiNand.dat only to use with bootctr. Sorry for not throughfully understanding. Yes, it's definitely not a bug.

hartmannaf, 2016-04-16:
And I normaly used the AuReinand.dat for testing commits, but now I see why it is failing, the .dat is only the fallback, I thought the main string will be different depending on the version(.bin/.dat).