LumaTeam/Luma3DS #36: (Enhancement) Patch to Allow Nintendo Zone to be launched

cheatfreak47, 2016-04-17:
I know this sounds odd, but the Nintendo Zone app is quite a bit larger than health and safety and there's proven work on it done that allows titles to be injected into it successfully.

http://gbatemp.net/threads/nintendo-zone-injection.423195/

As it stands more recent Home Menu's prevent the app from launching at all because "Nintendo Zone has been discontinued" error happens whenever you try to open the app, and it also happens when trying to resume the app if you launch it via something like Masher's Grid Launcher.

Just an idea, if that could patched, it'd be useful in the future for injecting stuff into clean NANDs
 ¯\(º_o)/¯

staticsn0w, 2016-04-17:
One of the Devs here, to be exact it is **4 **megabytes****! That space could go towards more useful things like a homebrew eshop or something, instead of a dead application.

piratesephiroth, 2016-04-17:
Whhich homebrew would you need to inject into a system app other than DevMenu?

piratesephiroth, 2016-04-17:
Anything else can just be packed into a CIA and installed normally afterwards.

staticsn0w, 2016-04-17:
@piratesephiroth no idea, but it gives devs a lot more freedom for their CIA installers.

Mrrraou, 2016-04-17:
I guess it would be nice, but not for everyone. My New3DS doesn't have the Nintendo Zone app, for example.

shinji257, 2016-04-17:
It might not show up until you actually get to a Nintendo Zone (like a Homepass relay or something).  It seems to stick around after it has shown once though.

ihaveamac, 2016-04-17:
Nintendo Zone always shows up now if you have it (since 3.0.0-5), so at this point they probably don't have the application at all.

staticsn0w, 2016-04-17:
@ihaveamac they still have the app. I used gridlauncher to bypass the HOME Menu warning, and injection works. n3ds just probably doesn't have NZone.

ihaveamac, 2016-04-17:
New3DS does have Nintendo Zone. some regions don't have it though I think (USA definitely does).

Mrrraou, 2016-04-18:
I have an EUR New3DS, which was 9.9 when bought, and downgraded to 9.2.

staticsn0w, 2016-04-18:
@ihaveamac @Mrrraou hmmm....try renaming your WiFi network to attwifi
that's what nzone searches for :p

staticsn0w, 2016-04-18:
if that doesn't work maybe @AuroraWright or someone who can navigate and understand the AuReiNAND code could patch that too.....

ihaveamac, 2016-04-18:
@Mrrraou out of curiosity, are you saying you don't have the icon on your Home Menu at all? if so, can you check if you have 0004001000022B00 / CTR-P-HMAP (from [3dbrew](https://www.3dbrew.org/wiki/Title_list#00040010_-_System_Applications)) installed to NAND?

Mrrraou, 2016-04-18:
Seems like it's installed, indeed. It's installed as "Visualiseur Nintendo Zone" (Nintendo Zone Viewer), with version 1034 and it is 4.58MB. However, I can't see it on the HOME Menu, but I guess it's not a big issue tbh. However, is Nintendo Zone injection necessary when we have a custom LOADER module ?

staticsn0w, 2016-04-18:
@Mrrraou launch it from gridlauncher, then exit it once it makes its extdata, then reboot. see if that works

staticsn0w, 2016-04-18:
Also, turns out USA v1034 is **~5.50MB**!

staticsn0w, 2016-04-18:
@Mrraou if that doesnt work ill make a HANS Nintendo Zone CIA in my spare time.

hartmannaf, 2016-04-18:
You should keep in mind that the nintendo zone is not discontinued in all regions, here in europe its still used for some thinks like codes for special events, from what I know.

hartmannaf, 2016-04-18:
@Mrrraou to be able to see the nintendo zone in your home menu, your 3ds needs to find a nintendo zone wifi at least one. For this its also possible to simply create an access point with a name like NintendoSpotPass2 .

staticsn0w, 2016-04-18:
@hartmannaf i thought you named the wifi access point attwifi....

hartmannaf, 2016-04-18:
@staticsn0w There are multiple name you can use, there is a list on [3dbrew](http://3dbrew.org/wiki/Nintendo_Zone#Hotspots)

staticsn0w, 2016-04-19:
@hartmannaf just scanned the AP name list. rename network to Bestbuy if you use WPA2-PSK, or attwifi if you use no passwd. [this is US ONLY. @Mrrraou just look at the 3dbrew entry to check the config you need]

hartmannaf, 2016-04-19:
Only the ssid is needed to lett it appear in the menu, unless you want to really use it. Also you can use every ssid, it will redirect you to your regions nintendo zone, if needed.

d0k3, 2016-04-19:
Don't know if this is the right place to discuss this, but here's my 5 cents. Sure, we can inject to the Nintendo Zone app, but should we? Wasted space is not an argument, cause technically you have a lot more wasted space on your CTRNAND / TWLN / TWLP which does not bother you that much. And, it is better to leave CTRNAND untouched as far as we can. 

Right now, FBI is small enough to be injected into any H&S app. With FBI injected, you can install anything else (and yes, you can install anything else into CTRNAND, but I don't recommend). Injecting into Nintendo Zone on top of that is pure cosmetics, and if it requires complicated patches, then @AuroraWright better works on something else instead of that.

If the Nintendo Zone icon bothers you, use FBI to uninstall (keep in mind it may still be useful, though).

RShadowhand, 2016-04-19:
attwifi or anything similar means nothing, as I've done homepass and it uses "attwifi" name. I still don't have Nintendo Zone app.

Shadowtrance, 2016-04-19:
Oh god what have i done....
The nintnedo zone injection was just a suggestion by someone on irc and me being bored so i did it. Atm i see no real need for it tbh. As @d0k3 said, FBI is currently still small enough to work with H&S (the smallest H&S app being CHN v5 o3ds - 604KB, largest being EUR v1 n3ds - 1016KB) so the extra space provided by other "useless apps" (looking at you Nintendo Zone, Face Raiders and AR Games, the latter 2 being just shy of 15MB each) isn't really needed at present.

RShadowhand, 2016-04-19:
I'm closing this, then.

shinji257, 2016-04-19:
Just an added note.  attwifi probably isn't triggering it because it isn't recognized as a Nintendo Zone by the 3DS anymore.  You might want to try NZ@McD1 instead if you want to really see it.

staticsn0w, 2016-04-19:
lol

staticsn0w, 2016-04-19:
only reason i suggested @cheatfreak47 to make this issue thread is because i wanted to give the testers a reason to test it, and to give the concept some functionality.

Gavin123-code, 2021-07-14:
I say if you allow a dead app to open actually make it have a point to be launched

Kadir-Wii-3DS, 2023-10-16:
https://github.com/Kadir-Wii-3DS/Releases-Nintendo-Zone-Viewer-CIA/releases take this GUYY

closed 2023-10-16 by PabloMK7