LumaTeam/Luma3DS #38: Clash in new payloads

habbbe, 2016-04-17:
Hi!
I just checked your latest release. I can't try it out right now, but from looking at the code, it seems that the new pattern used for allowing the old payloads to be recognized will also produce a clash between the payload "r" and "right", as r*.bin will also match right.bin. That's why I had the somewhat hard on the eyes solution in my initial commit that checked both the original name, and also the extended pattern with _NAME.

evrardjp, 2016-04-17:
Hello. I should take the blame on this one.
Would that be fine if R1 would be loaded by r1 instead of only r? 
Or should we rollback? I'll be happy to send a PR. R1 makes more sense to me.

AuroraWright, 2016-04-17:
This won't happen because of the order. "right" is checked before "r".

habbbe, 2016-04-17:
But the right is only checked if the right button is pressed, thus if the user holds down R, it would have already passed the right-condition by failing the (pressed & BUTTON_RIGHT), and it will load the right-payload anyway.

AuroraWright, 2016-04-17:
OK, reverted back to 5.1. I suppose renaming one or few files won't be much work (I only removed the _ because I thought it was a quick way to also preserve compatibility, but I don't really care).

closed 2016-04-17 by AuroraWright