LumaTeam/Luma3DS #39: Smash bros (eshop) Not working. [Black screen]

BOBdotEXE, 2016-04-18:
even since 5.0, smashbros boots to a black screen,
I've tired toggling some settings, and had no luck.
It's officially downloaded, and purchased from the e-shop, and I even tried re-downloading.
I know the game takes longer to load then others, but even after 10+ mins, it will still not have loaded.

[Emunand updated to the latest version]
AureiNand 5.1.1 +a9lh

AuroraWright, 2016-04-18:
Is it too hard to read the wiki?

closed 2016-04-18 by AuroraWright

BOBdotEXE, 2016-04-18:
Considering I did not know there was a change that would even require a different setup,
No, I did not check the section that had a heading called "Fixing GBA/DSiWare issues"
For my issue with a non-GBA/dsiware game.
 
But still, thanks for your work on the project, I'll use the process noted there.