LumaTeam/Luma3DS #42: Should I use SFN(8.3) format in the payloads name?

Asellus, 2016-04-20:
> v5.1.1 
> - Fixed the L+SELECT payload not working, this was due to the LFN -> SFN (8.3) conversion. Now the payload must be named sel_NAME.bin.
> - Removed LFN support from the FatFs of the chainloader (as LFNs aren't used at all by it). This shrinks down the binary by ~4 KBs.
> 
> Note, you don't need to update if you didn't use the L+SELECT payload or if you aren't obsessed by the binary's size :)
>
> https://github.com/AuroraWright/AuReiNand/releases/tag/v5.1.1
> https://github.com/AuroraWright/AuReiNand/commit/a11124a3d60019ca572330b591ca65371a0a2eb7

(e.g. )

| Button     | payloads     |
|:----------:|:------------:|
| ↑         | up_xxxxx.bin |
| ↓         | down_xxx.bin |
| ←         | left_xxx.bin |
| →         | right_xx.bin |
| X          | x_xxxxxx.bin |
| Y          | y_xxxxxx.bin |
| L+R        | r_xxxxxx.bin |
| L+SELECT   | sel_xxxx.bin |
| START      | def_xxxx.bin |

Am I wrong ?

digitaldiatribe, 2016-04-20:
Yes.

Asellus, 2016-04-20:
If yes ...
I think  

FILE NAME MUST BE 12 CHARACTERS OR LESS!  [SFN(8.3) format]

On the  [Installation and Upgrade page](https://github.com/AuroraWright/AuReiNand/wiki/Installation-and-Upgrade)
"Upgrading to 5.1.1 from 4.3 or older" & "Upgrading to 5.1.1 from 5.0"

On the [Other Features page](https://github.com/AuroraWright/AuReiNand/wiki/Other-features-and-notes#payload-chainloading) 
"Payload Chainloading"

AuroraWright, 2016-04-20:
No you shouldn't. If you have a file named x_whateverlongfilename.bin its SFN is something like X_WHAT~1.BIN, I only need the SFN to have X_ to detect it. Same for the other payloads

closed 2016-04-20 by AuroraWright