LumaTeam/Luma3DS #45: X Button Payload not loading in Version 5.1.2

DonDalle, 2016-04-20:
For me the X payload won't load up in Version 5.1.2, I get a black screen when I want to boot into it. It works fine in 5.1 though

Thanks for your great work!

RShadowhand, 2016-04-20:
What is your x payload? Emunand9?

DonDalle, 2016-04-20:
it is Uncart. 

My payloads are like this: 
[image]

RShadowhand, 2016-04-20:
There was an issue with D9 and E9 that required manual patches on ARN's side, but those were fixed by d0k3. It might be that Uncart also needs to be updated to work.

DonDalle, 2016-04-20:
ok thanks!

RShadowhand, 2016-04-20:
To troubleshoot, you can swap x and y payloads, see if you can boot with x then. If you can, then the problem is uncart itself.

DonDalle, 2016-04-20:
I can do it later and will report back.

DonDalle, 2016-04-20:
The problem is with Uncart. I swapped Uncart (X) with Emunand (Y) and starting Uncart with Y also doesn't work.

closed 2016-04-20 by DonDalle