LumaTeam/Luma3DS #50: DNS Blocking?

AP0-X, 2016-04-22:
I read that you block updates and I know that the old way to block push updates was to use the tubehax DNS. Now the issue is when I try and go online in ORAS everything works but battlespot. It gives error code 006-1102 which is a connection error and it recommends going through googles DNS if experiencing problems. This however doesn't work and I have tried for about a week (Switching between Auto and Google DNS) to resolve this issue. Then I read the changelog and thought that you might be blocking all of Nintendo's services other than eshop and NNID and that is how you achieve no push updates. I have tried my cartridge and CIA versions(Made by Uncart and FunKey) and neither worked and I have restarted my router and power cycled it multiple times. I am running AuReiNand 5.1.2 on an Original 3DS.

TuxSH, 2016-04-22:
I heard that Tubehax doesn't really block 3DS updates anymore, anyways.

RShadowhand, 2016-04-22:
Just for testing, try SaltFW, see if that fixes your problem, if it does,
tell AuroraWright to accept my PR. :P

hartmannaf, 2016-04-22:
Have you checked your dns settings?
None of the patches should make any problem, since the system module, which is checking for updates, is patched to return that no update is available.
On wich firmware version you are on?

AP0-X, 2016-04-22:
Update responses(thanks everyone)

Hartmannaf

I'm on 1.7 eur and have tried many DNS settings so this post didn't seem like a regular FAQ mistake.

TuxSH
I'm just speaking of past experience when it used to be tubehax or ninjhax not even browserhax was discovered when I remember this. I just though that might be the cause.

Rshadowhand 
I'll try it within the next day, salt just came out for a9 as far as I know and so haven't tried it yet. I have tried cakes however.

RShadowhand, 2016-04-23:
@AP0-X If cakes didn't work, then you have a deeper problem than the device itself. No need to try SaltFW in this case.

closed 2016-04-27 by AP0-X