LumaTeam/Luma3DS #52: [Request] Make SysNAND default instead of EmuNAND

Margen67, 2016-04-22:
Since EmuNAND is obsolete and no longer needed with the release of A9LH, I believe that SysNAND should be the default option instead of EmuNAND.

shinji257, 2016-04-23:
Luma3DS isn't used exclusively by A9LH users.  It is used by many Menuhax users as well and as such the sane defaults needs to consider those users as well.  For this reason EmuNAND should remain the default.  On first boot you have the option to enable Updated SysNAND.  Most people know if they have an EmuNAND or not.

Not entirely certain but I'm pretty sure if EmuNAND is missing for some reason it will fallback on SysNAND anyways.

Margen67, 2016-04-23:
> Luma3DS isn't used exclusively by A9LH users.
Yes, but the amount is growing.

> It is used by many Menuhax users as well and as such the sane defaults needs to consider those
> users as well. For this reason EmuNAND should remain the default.
> On first boot you have the option to enable Updated SysNAND.
Are you implying SysNAND is insane?
Seeing as how Menuhax is inferior EmuNAND shouldn't be the default.
Updated SysNAND is the new standard.

>Most people know if they have an EmuNAND or not.
What does that have to do with it?

shinji257, 2016-04-23:
You obviously do not know what "sane defaults" are.

Sane defaults are a set of settings that will apply properly to the majority of users.  It doesn't imply any other setup is insane.

Just an FYI.  I'm an "Updated Sysnand" user here however I also believe that any default setup should apply to as many users as possible.  Not just my own individual setup.

CecilFF4, 2016-04-23:
I will never stop using emunand for one reason: easy backup. I plug my SD in my computer and I got a whole emunand backup in one minute. I've had to restore said backup many times when I mess around with system files and whatnot. Emunand is most certainly NOT obsolete.

Oh, and if you want sysnand by default, just select that option in the config. Why is there even a request for it?

Margen67, 2016-04-23:
>Emunand is most certainly NOT obsolete.
Only in specific cases like yours, or mine.
I have to use a 9.2 EmuNAND to boot Gateway into my SysNAND until they add A9LH support.
But for a user that just wants to install and play games there's no need for an EmuNAND unless they're using something like Menuhax, which in of itself is obsolete unless you're paranoid about bricking.

TuxSH, 2016-04-23:
If no emuNAND exists, Luma3DS will patch and boot sysNAND anyways.

closed 2016-04-23 by TuxSH