LumaTeam/Luma3DS #54: Potential for patch to block decrementing demo plays?

vaguerant, 2016-04-23:
Is there any chance we could get a patch to launch demo titles without decrementing the number of plays remaining or otherwise bypassing the limited-play system? Some demos would particularly benefit from this, e.g. Bravely Default's demo features content that is unique to the demo, not a subset of the retail game. Being able to play this indefinitely (which is, admittedly, already possible via HANS), would improve it greatly, in addition to being more convenient and, probably, stable than using HANS to do the same. There are many titles that currently don't cooperate with HANS, including a large proportion of demos.

According to @yifanlu: "I think it should be easy. likely a home menu patch somewhere"

TuxSH, 2016-04-23:
Atm I've no idea about which service commands are used to store the updated count, something I'd like to know.

poutros, 2016-04-23:
I think lauching through homebrew doesn't decrement the demo plays.

vaguerant, 2016-04-23:
I think that depends on how they're launched. As far as I know, launching titles via RegionFour does, but via HANS doesn't. Obviously they operate in pretty significantly different ways, but I'm not sure how much help that is for someone looking to do it directly in CFW.

TuxSH, 2016-04-23:
Yes of course, it's HomeMenu who decreases the counter. But how exactly?

RShadowhand, 2016-04-23:
The demo counter is held in the header. It's decreased by HomeMenu each
time it's launched through HomeMenu.

TuxSH, 2016-04-23:
Of course I knew that (3dbrew), but how precisely is it done... ?

RShadowhand, 2016-04-23:
That is up for research. I just know that it's counted in the header, and HomeMenu decreases it each time you run it. Which is why HANS can launch it without decreasing the numbers.

TuxSH, 2016-04-23:
Yes of course ...
But I'll be busy the next two weeks...

rboninsegna, 2016-04-26:
FWIW, I can delete the AM driver's save without ill effects.

I don't have an unpatched demo I could try it on right now, though!

llakssz, 2016-05-27:
I played with demo counts last year I think, patching the ticket and deleting the correct system save both work.
Although in a weird case I found (and another user too) - when the ticket was patched, sometimes (rarely) the correct demo count would return...

But yes this would be useful and nice to have in cfw!

hartmannaf, 2016-05-27:
@llakssz The eshop resets the tickets you own on start, so the counter will return.

HugaShire, 2017-02-07:
Use homebrew, also there is a tool for resetting play count

closed 2017-02-09 by TuxSH