LumaTeam/Luma3DS #56: GPL violation

Reisyukaku, 2016-04-23:
Removing the copyrights from the files was a violation of GPL. Please revert changes appropriately.

GPL, art.4(1):
"You may convey verbatim copies of the Program's source code as you
receive it, in any medium, provided that you conspicuously and
appropriately publish on each copy an appropriate copyright notice;
keep intact all notices stating that this License and any
non-permissive terms added in accord with section 7 apply to the code;
keep intact all notices of the absence of any warranty; and give all
recipients a copy of this License along with the Program."

https://github.com/Reisyukaku/ReiNand/blob/master/LICENSE.txt#L195

fox8091, 2016-04-23:
You can't slap copywright on files that aren't yours to begin with. The only one REMOTELY copywritable was types.h

Reisyukaku, 2016-04-23:
The ones without copyright weren't mine, the rest were obviously. Also please prove they weren't mine.

AtrusHB, 2016-04-23:
"Please prove they weren't mine."
[That'sNotHowItWorksYouLittleShit.png](http://wiki.magicalgirlnoir.com/images/7/70/That's_not_how_it_works_you_little_shit.png)

Reisyukaku, 2016-04-23:
My point is, they were removed, and thats against the terms. Nothing more needs to said. The proof is in the commits.

TuxSH, 2016-04-23:
Except for ```emunand.c```, all of the other files have been written or completely rewritten, some others came from other projects than ReiNAND.

```crypto.c``` originally comes from http://github.com/b1l1s/ctr .

Some other files (like ```types.h``` and ```memory.h/.c```) are simple enough that it is impossible to prove it's you're own. A lot of people could have written that code / you can find that code "anywhere".

AtrusHB, 2016-04-23:
"The proof is in the commits."
I don't think you'd be liking your own posts if they actually proved anything.

Reisyukaku, 2016-04-23:
@TuxSH The only things i didnt write were crypto.c, fatfs/sdmmc, and the linker and other pedantic things. And as such, i didnt put my name on it. Everything else is purely mine.

ryanrocks462, 2016-04-23:
.

RShadowhand, 2016-04-23:
I think you're just being an asshole for the sake of being one.

TuxSH, 2016-04-23:
https://github.com/mid-kid/CakesForeveryWan/blob/master/source/fs.c
https://github.com/Reisyukaku/ReiNand/blob/fd2d29bb006d723b2db31b12061a3442560fe83d/source/fs.c

https://github.com/mid-kid/CakesForeveryWan/blob/557a8e8605ab3ee173af6497486e8f22c261d0e2/source/memfuncs.c
https://github.com/Reisyukaku/ReiNand/blob/fd2d29bb006d723b2db31b12061a3442560fe83d/source/memory.c

In these files you just copied @mid-kid's code (for at least some functions), just changing variable names; those files are license under the top-level LICENSE.txt. Where are your credits, @Reisyukaku ? Not only that, but this techinally makes your post a lie.

Reisyukaku, 2016-04-23:
Nice using an example with my name not in it. As for memory one, only one worse that you could have used was types.h, lol.

AuroraWright, 2016-04-23:
How exactly does that clause apply here?
"keep intact all notices stating that this License and any
non-permissive terms added in accord with section 7 apply to the code" and you had exactly zero statements stating that the code was GPLed or was under the GPL.

TuxSH, 2016-04-23:
@Reisyukaku http://www.gnu.org/licenses/gpl-faq.html#GPLOmitPreamble

ryanrocks462, 2016-04-23:
tfw one of you autistic niggers deletes my comment qq

OctopusRift, 2016-04-23:
@ryanrocks462 dude. gtfo.

ryanrocks462, 2016-04-23:
@OctopusRift stopped being a faggot and trying to defend this clearly stolen code

TuxSH, 2016-04-23:
No spam allowed.

OctopusRift, 2016-04-23:
@ryanrocks462 go read a book on GPL

Reisyukaku, 2016-04-23:
Ok this is getting carried away. I'm too old to deal with a bunch of children.

closed 2016-04-23 by Reisyukaku

ryanrocks462, 2016-04-23:
@OctopusRift be smart and figure out when GPL is being stated in a commit just not directly =)

OctopusRift, 2016-04-23:
@ryanrocks462 this is closed.

TuxSH, 2016-04-23:
@Reisyukaku "keep intact all notices stating that this License and any
non-permissive terms added in accord with section 7 apply to the code;"

`c
/*
*   fc.c
*       by Reisyukaku
*   Copyright (c) 2015 All Rights Reserved
*/`

Where in earth does this notice says the code is under the GPL license, tell me?

ryanrocks462, 2016-04-23:
@OctopusRift obviously

astronautlevel2, 2016-04-23:
Also, putting a file in the root of the project isn't enough to signify code is under the GPL: http://www.gnu.org/licenses/gpl-faq.html#LicenseCopyOnly

ryanrocks462, 2016-04-23:
uh oh both of your autism is showing again :/

ryanrocks462, 2016-04-23:
when fuckers can't handle the hate so they just remove your comments kek,  ¯\_(ツ)_/¯

mid-kid, 2016-04-23:
Yay, GPL stuff. Sorry for piping in, shouldn't have highlighted me if you didn't want me to.  
I can't see where Reisyukaku's copyright is removed, except for `types.h`, but why would you enforce GPL on some type definitions that are pretty universal?  
But I guess I'm just repeating what everybody's already said. I'd honestly like @Reisyukaku clarificating a bit more instead of ragequitting like this.

TuxSH, 2016-04-23:
You should add proper disclaimers to your (and your collaborators') files btw @mid-kid.

Reisyukaku, 2016-04-23:
@mid-kid I'm not rage quiting, It's just not worth my time lol. I have shit to get done. 
As for your question, i wasnt too concerned about little thinks like types.h really.. basically everything else.

mid-kid, 2016-04-23:
I indeed should, but I've always disliked having a huge comment at the top of each file. I just expect people to respect the license in either case (and add proper credit if copying it). It's not like I can prevent someone from removing the header if they wanted to (except calling them out for it).
Though you're right, I should, and I probably will in some commit soon.

OctopusRift, 2016-04-23:
@mid-kid is a true gentleman.

mid-kid, 2016-04-23:
@Reisyukaku:
>basically everything else.

Please point out what. I've checked some commits but didn't find anything, and I don't expect the regular passerby will go through all the commits before judging.

Reisyukaku, 2016-04-23:
"It's not like I can prevent someone from removing the header if they wanted to (except calling them out for it)."

My thoughts exactly. But it's turned into a circus instead, lol

TuxSH, 2016-04-23:
Still liking your own posts @Reisyukaku ?

ryanrocks462, 2016-04-23:
still being autistic? @TuxSH =)

astronautlevel2, 2016-04-23:
Had you even added a line in the readme signifying that the code was under the GPL license there would be a lot less room for debate here, but nowhere in your entire project is there any indication of licensing for the code, with the exclusion of license.txt. As I have already pointed out, that's not enough to signify the code is under GPL licensing.

ryanrocks462, 2016-04-23:
i love how 
@TuxSH commented 24 minutes ago
>No spam allowed.

@TuxSH commented a minute ago
>>>>Still liking your own posts @Reisyukaku ?

k, fucking fag -.-

ThievesNeverProsper, 2016-04-23:
thieving scumbags should respect the maker

TuxSH, 2016-04-23:
As @astronautlevel2 said, you could at least add one line saying it's under the GPL.

mid-kid, 2016-04-23:
Okay, I understand why @Reisyukaku wanted to leave.  
Comments like
>Still liking your own posts @Reisyukaku ?

And whatever ryanrocks is posting are completely inappropiate.
Count me out as well, if we can't keep this civil.

TuxSH, 2016-04-23:
Removed the comment.

ThievesNeverProsper, 2016-04-23:
give the man his credits you thieving scum buckets

AuroraWright, 2016-04-23:
@ryanrocks462 What does autism have to do with anything? What if I told you I were myself?

Youngsie1997, 2016-04-23:
TIL that "Watching" github repos is a bad idea

ryanrocks462, 2016-04-23:
@mid-kid trust rei loves my comments ;)
and @AuroraWright you would just prove my point =)

ThievesNeverProsper, 2016-04-23:
give Rei him credits & stop stealing his code !!!!!!!!

Youngsie1997, 2016-04-23:
I'm all for a pissing contest and slagging match if it actually leads some where productive but its quite clear this isn't going anywhere soon so im out lol

ThievesNeverProsper, 2016-04-23:
GIVE REI HIS CREDITS YOU CODE STEALER

ryanrocks462, 2016-04-23:
@TuxSH their wasn't anything even basin my last comment and your removed it, lmfao your fucking ASS-Blasted Right now

mid-kid, 2016-04-23:
>Removed the said comment.

Remivng a comment is useless when the damage is already done. I personally hate censorship, as it only helps adding more confusion (and those damn reaction icons github added recently).
Requesting anyone with privileges to lock the thread.

TuxSH, 2016-04-23:
Ok, sorry for that.
It wasn't my intention.

Anyways, this is getting toxic/spammed, so I'm locking.

ThievesNeverProsper, 2016-04-23:
!!!!!!!!! CODE STEALING SCUM !!!!!!