LumaTeam/Luma3DS #58: EmuNAND 10.5 on SysNAND 4.3

julianxhokaxhiu, 2016-04-25:
Hi,

I managed to boot your CFW on my O3DS through MSET, although when I try to launch the CFW a black screen is there, and the EmuNAND is not loading. With rxTools everything boots fine.

I tryed to place the `firmware.bin` file in `luma/firmware.bin` but it doesn't work anyway.

Any clue? Am I missing a step?

Thanks in advance,
Julian

ihaveamac, 2016-04-25:
I'm pretty sure Luma3DS only works on consoles with 7.0 sysnand or higher, or arm9loaderhax.

julianxhokaxhiu, 2016-04-25:
Just in case, would it really be difficult to support 4.x FWs?

ihaveamac, 2016-04-25:
I'm not a contributor to Luma3DS but I think it's common practice now to just update SysNAND to 9.2. nobody really supports staying behind on 4.x anymore.

AuroraWright, 2016-04-25:
As @ihaveamac said, this doesn't support anything lower than 7.x when using *hax (on A9LH, in theory all FIRMs should work, but the patch aren't tested and I don't recommend it at all).

closed 2016-04-25 by AuroraWright