LumaTeam/Luma3DS #59: Backlight control

poutros, 2016-04-26:
It is already possible to control it for payloads but what about the home menu/games? I know it is possible because ntr does it but it doesn't stick after the lid is closed and reopenned. Would it be possible to implement a backlight control that goes below what the 3ds allows? It would be enabled as normal on the luma3ds menu but it would be separate for payloads/luma menu and normal 3ds so that we are able to change it if we set it too low.

RShadowhand, 2016-04-26:
Not possible, because once the home menu is initialized, it sets its own
brightness, so setting it at a9lh won't do anything.

poutros, 2016-04-26:
So it has to be set after the home menu boots? What about games? Or does the home menu reset the brightness on game booting?

RShadowhand, 2016-04-27:
Yes, it has to be set after the home menu boots, which isn't possible with Luma. As far as I know, home menu sets the brightness level every time it inits the screen, which is why NTR's brightness doesn't stick. Closing, because it's not possible as it stands.

closed 2016-04-27 by RShadowhand

hartmannaf, 2016-04-27:
for TWL/GBA you can set the Brigthness after pressing home using UP/Down+STart+Select if I remember it correctly.

poutros, 2016-04-27:
But couldn't luma block the home menu from trying to set the brightness everytime? That way it would probably be able to set it itself, right?

RShadowhand, 2016-04-27:
Don't quote me on this, but as far as I know, no.