LumaTeam/Luma3DS #61: 5.3.1 blocks DStwo NDS flashcart

tobeannouncd, 2016-04-27:
I tried out 5.3.1 today on my O3DS, and I have had no issues other than my DStwo DS flashcart no longer working. Everything seems to work fine on 5.3, but I have not been able to boot into the DStwo on 5.3.1.

tranaconda, 2016-04-27:
Working for me on O3DS and N3DS

Hellrespawn, 2016-04-28:
Having no problems with my DSTwo on N3DS, 5.3.1.

tobeannouncd, 2016-04-28:
Yeah, I'm not using any boot managers, just plain old A9LH, so pathchanger won't do me any good (I tried it for kicks, but it didn't make any difference). Still doesn't explain why 5.3 works and 5.3.1 doesn't.

RShadowhand, 2016-04-28:
As far as I can see, there hasn't been a change on that patch from then, so I don't really know.

tobeannouncd, 2016-04-28:
I did some more testing, and it seems like other DS games don't boot, but strangely enough, Smash boots.

cheatfreak47, 2016-04-28:
@tobeannouncd 
Perhaps your SysNAND's TWL FIRM is busted?

Try redownloading the title with 3DNUS and installing/replacing the title with FBI.

I believe it is
0004013800000102 v8817 for o3DS/2DS
0004013820000102 v9936 for New 3DS

Install the one appropriate for your 3DS's NAND using FBI and it should prompt you to overwrite it or something along those lines.

tobeannouncd, 2016-04-28:
@cheatfreak47 
That totally worked! I'm still a bit curious why it didn't cause issues with earlier versions of Luma3DS, but I'm just thrilled that it's working now!

closed 2016-04-28 by tobeannouncd