LumaTeam/Luma3DS #67: Implement "Shared Font Tool" functionality

chaorace, 2016-04-30:
Feature request on the behalf of /u/youjun540164

As I understand it, Chinese 3DS users need to use a tool called [Shared Font Tool](http://pan.baidu.com/share/link?shareid=3005947731&uk=1882167608) to play most out of region games without garbled characters. I'm sure a lot of our friends in the chinese gaming community would be very pleased if this feature could be implemented on the CFW level!

For more info, see the thread [here](https://www.reddit.com/r/3dshacks/comments/4h2pcs/luma3ds_54_release/d2my1wu)

AuroraWright, 2016-04-30:
I honestly have no idea what this does... I can't find source code for it, so I assume it's closed source? Any details on what it actually does?

chaorace, 2016-04-30:
I'm not entirely sure myself, but, based on the contents of the zip, my best guess is that it swaps out the loaded fonts in memory with a given bcfnt file. Then again, I don't know very much at all about how the 3DS handles fonts! I can ask the original requester, who speaks Chinese, if the devs could elaborate on how the app works / release source

piratesephiroth, 2016-04-30:
Yeah, by reading the description I guess it changes the home meu region and loads the different font file

AuroraWright, 2016-04-30:
I'd need to know how this works to be sure... What I'm sure about is that a game-specific toggle is not possible, due to how "loader" works (same reason for which per-game CPU clock can't be done). If this patches a system title/module (as I'm guessing), the patch would be there for all the booted games until you change it... so maybe this would be pointless.

piratesephiroth, 2016-05-01:
I think there's also the additional problem of having to use the copyrighted font data.

zestybaby, 2016-05-01:
There's open source CJK fonts from big company so that wouldn't be a problem, the only problem is we are not sure if the original dev would corporate with this project. As far as I know from 3DS scene in China, there's only a tiny chance that they would release the source code to public.

fincs, 2016-05-01:
Why not politely ask them for source, expressing the intention to integrate it into a CFW for the benefit of Chinese users?

giveup, 2016-05-02:
Here is a tool that convert font into 3ds format font.
 https://github.com/dnasdw/3dsfont
 J/U/E region 3ds use standard font that not include all characters.

dnasdw, 2016-05-03:
https://github.com/dnasdw/SharedFontTool

zestybaby, 2016-05-04:
Can confirm the user above is the original coder for that tool. It's really glad to see Chinese devs step up and release source code.

paging @AuroraWright

llakssz, 2016-05-27:
@chaorace Nice idea, but not that important like you say. Most Chinese users don't use Chinese 3DSs, they use Japanese 3DSs, the real Chinese systems are quite limited.
Nevertheless still a good idea, but I don't think it's that big a need.

Many people use the Shared Font Tool the other way around, to enable the Chinese font on their other systems, when playing Chinese translated games on their Japanese 3DS, etc.

closed 2016-11-06 by TuxSH

d3m3vilurr, 2016-12-02:
@TuxSH no more planed this feature? localized game in asia region still need this feature.