LumaTeam/Luma3DS #68: Patch out requirement to access conntest.nintendowifi.net

tranaconda, 2016-04-30:
I have blocked one of my 3DS from WAN for testing reasons and noticed that all apps that use WiFi are stuck "Waiting for wifi..." (FTPd) or something unless they are allowed to connect to conntest.nintendowifi.net. This means you can't use network apps in closed networks or when your internet is out.

I did get arround this by poisoning my local DNS to point to my computer and serving the webpage there, but it'd be nice to patch this out somehow (optionally, as I realize it could break other things).

Browsing 3dbrew.org, I guess it must be somewhere in ac:u, probably GetWifiStatus or IsConnected.

PokeAcer549, 2016-05-09:
From my knowledge, it's part of the WiFi code itself, as it is on Nintendo DS, DSi, Wii, WiiU, etc..

tranaconda, 2016-05-09:
Please be more specific, what exactly do you mean by "WiFi code". Is my assumption of it being in ac:u wrong?

d0k3, 2016-05-09:
This is correct - this is part of the Wifi code. I don't know if Aurora can fix this in Luma3DS, and also, if it is a good idea to do this, cause apparently the console needs some way to check if Wifi is working.

RShadowhand, 2016-05-09:
Your best bet would be to redirect conntest.nintendowifi.net to a local address within the network using your router, or simply using a DNS server that does that.

tranaconda, 2016-05-09:
If you read the OP, you can see thats what I'm doing right now. But this is annoying and people who don't know about DNS poisoning can't use FTPd or other WiFi enabled apps when their internet is down. It could be an optional patch, as I suggested.

piratesephiroth, 2016-05-09:
MicroSD Management doesn't require the connection test to be successful (it can be used in a local network without internet access) so it's definitely possible.

TuxSH, 2017-05-08:
The thing is that conntest.nintendowifi.net is used to automatically select which access point to use.

It doesn't apply to MicroSD Management since the latter forces you to select a precise access point.

closed 2017-05-08 by TuxSH