LumaTeam/Luma3DS #70: Implement granular CPU control similar to language emulation

lpchaim, 2016-04-30:
This is pretty much the only thing that bothers me with luma3ds right now, as neat as the concept itself is we can only change CPU settings on the boot menu and it's permanent. Some games don't benefit much or at all from enhanced CPU clock/L2 cache to warrant a permanent change and rebooting to change is a tad too cumbersome. A system akin to how the language emulation feature currently  works would be ideal IMO, as in a text file with the game id as a title placed in /luma/cpu/ and its contents being worded the same as in the config menu, "Off", "Clock", "L2" or "Clock+L2".
Sorry if this sounds unfeasible or unreasonable, I have no real experience in this kind of programming to speak of.

cheatfreak47, 2016-04-30:
This is a duplicate of the Enhancement I brought up in #25 

Has anyone over there at least looked into what it would take to implement?
I understand that as it stands the current patch would need to be scrapped entirely in favor of a new method, but I still think it'd be worth the effort.

Would definitely set Luma3DS above and beyond all the other CFW on the 3DS, it has a huge effect on battery drain as it stands, and one of my games (Sonic and All-Stars Racing Transformed) crashes with the n3DS Clock Speed when the vehicles transform in-game.

RShadowhand, 2016-04-30:
It's not very possible as it stands, because it needs more research. As far as I've done, the headers for the game has to be patched on the fly, which isn't possible with 3ds_injector.

lpchaim, 2016-04-30:
I see. So basically something like NTR's implementation doesn't help at all like I uneducatedly assumed. Well then, I suppose this issue doesn't really have anything to stand on ATM so feel free to close this.

RShadowhand, 2016-04-30:
Closing until further research is conducted.

closed 2016-04-30 by RShadowhand

AuroraWright, 2016-04-30:
This is probably impossible within 3ds_injector.