LumaTeam/Luma3DS #72: Level 5 Brightness

reworks, 2016-05-02:
I'm not sure if its the CFW or my setup, but when I set the brightness to 4, the highest in the config menu, the screen is still rather dull. I was wondering if we could get a level 5 brightness? I think the offset is 0xFF to make it the same as the 3ds's level 5.

arm9loaderhax: Delebile's Fork
Installation: I followed Plailect's guide.
Console: 2DS v10.7 Updated SysNand
Luma3DS: v5.4

Terathiel, 2016-05-02:
It's your setup. Luma3DS's brightness setting is only applicable when using a no-screeninit fork of A9LH, such as Aurora's own ([see here](https://github.com/AuroraWright/Luma3DS/wiki/Options-and-usage)). delebile's fork already initializes the screen, and as such the Luma3DS setting does nothing.

IIRC brightness 4 is already equivalent to home menu 5. Not sure if there's any way to adjust the brightness using delebile's fork.

reworks, 2016-05-02:
Problem solved, I guess. Plailect's guide should probably be updated to reflect this, but this isn't the place to discuss it.

closed 2016-05-02 by reworks