LumaTeam/Luma3DS #73: Revert SOAP patch removal?

execfera, 2016-05-02:
Is it possible to restore the XML country response patch in NIM so that we can have a proper eshop-compatible region-change? It would be beneficial to those who don't mind ditching their old NNID, or for people who bought a JP non-XL N3DS (like me) who wish to change their region permanently.

I don't think it's too far of a stretch to ask for, given that Plailect's guide already has a region change section. It could be implemented such that someone could dump their own SecureInfo_A, change the region byte to their target region, and put it in /luma or some such.

AuroraWright, 2016-05-02:
You can't fully region change (that is, have a fully working eShop - including... well, buying/downloading stuff) without following this https://gist.github.com/yifanlu/e80db121d38aceb8cca0e03cefd5853b/3c4dd89869156ca0f945a2791e699acfdb32b510 . I tested this on my New 3DS, I have a perfectly region changed NAND, I ditched my NNID for the test, and there's no way to have downloads/purchases working. You can just browse the eShop (which I can also do without that patch). As it stands, that patch is useless for 99,99999% of people (that guide is really far from being user friendly)

closed 2016-05-02 by AuroraWright

piratesephiroth, 2016-05-03:
you'd have to change the serial too.