LumaTeam/Luma3DS #75: Experiencing lag when playing Smash local multiplayer.

Lykie, 2016-05-04:
Exactly what the title says. Seems to be getting worse, maybe with each Luma release. Can't for sure say it's Luma's fault but it didn't happen so badly before. The lag slows the game down to a few frames on each 3DS before correcting itself.

cheatfreak47, 2016-05-05:
This is very unlikely to be a problem with Luma 3DS. I've played Smash both online and locally between 4 3DS systems all but one running Luma, and no such issue happens.

The area you are in has a lot to do with something like that and there's a variety of things that can cause local network interference.

RShadowhand, 2016-05-07:
As @cheatfreak47 mentioned, this has nothing to do with your CFW.
Have you tried playing with Cakes or SaltFW?

Lykie, 2016-05-07:
@RShadowhand I have not to play used saltfw. I only have it on another sd card that has basically minimum space on it, no room for smash update.

RShadowhand, 2016-05-16:
Closing because it's not related to what CFW you use.

closed 2016-05-16 by RShadowhand