LumaTeam/Luma3DS #78: I'm not able to load RedNAND

SuryaKmuar, 2016-05-07:
I followed till step 23 of Part 3 (RedNAND) .I followed every step correctly but when i try to boot RedNAND using Luma3DS by first activating "show current NAND in system settings" then pressing start, it boots to home menu but when i go to system settings, I do not see 'Emu' in front of system version. What should i do?Is there any way to troubleshoot this problem? 
My downgrade was successful, no partial downgrade.

Shadow322, 2016-05-07:
U shitting me?

RShadowhand, 2016-05-07:
Delete `luma/config.bin`, and try again.

SuryaKmuar, 2016-05-07:
Actually it worked, Important announcement  The latest version of luma3ds v5.4 has a bug. For some people it will not boot into RedNAND because this bug occurred to me, so i used old v5.3.1 of luma 3ds and it worked! I enter RedNAND without any problem.I recommend to all to use v5.3.1 instead of v5.4 if you face this problem after using v5.4 of luma3ds.

closed 2016-05-07 by RShadowhand