LumaTeam/Luma3DS #82: Handling non-genuine N3DS headers / TWL MBR

d0k3, 2016-05-09:
This issue may pop up on your radar sooner or later anyways, so I figured it may be best if you get the description from me.

NAND headers are not identical between consoles (even if they are of the same type), they contain the slot 0x03 (console-unique) encrypted MBR for the TWL partitions starting at 0x1BE. See here:
https://www.3dbrew.org/wiki/NCSD#NCSD_header

Now, there are those people who lost their original headers (in the downgrade process, forgetting to do a backup before), and those have to use another consoles NAND header. Fixing another header for a console is not possible due to the NCSD header signature, which (afaik) is checked even before A9LH gets into action. With an unfixed, nongenuine header, TWL does not work (due to a bad MBR). The decrypted MBR, however, is the same for each and every 3DS console out there (identical between O3DS / N3DS, too).

Now, I know, only a few isolated cases can benefit from fxing this in Luma3DS, and this may be too much work for that. So, feel free to close without fixing :):

TuxSH, 2016-05-09:
If I understand correctly, we can't fix bad MBR partitions or it causes a brick?

Since DSiWares run on the bar metal and read/write from/to NAND directly, maybe they read the MBR too. (or maybe they don't, I don't really know)

d0k3, 2016-05-09:
Yup. Fixing and reencrypting the MBR is easy, fixing the signature is impossible. You cannot alter the MBR table, cause any altered byte in the header results in a blackscreen brick. We tried (meaning: @al3x10m tested it).

TuxSH, 2016-05-09:
A possible solution (and maybe it's the only one) would be to reencrypt the TWL partitions using the value of the TWL keyslots from the source console, and having Luma (or anything else) set the keyslots accordingly (on the ITCM and on the AES engine)

d0k3, 2016-05-09:
Although this is pretty much possible to do, it would further remove the console from a genuine state. It would also require additional steps beyound Luma3DS (reencrypting TWL stuff), and: noobs. Taking into regard that this is good for isolated cases only (who are at fault for that problem in the first place), I'm a bit averse to that solution.

So... the NAND header signature can't be patched, my assumption for that is right, correct? TWL running on bare metal also means, in essence, that we'd need a CFW running in TWL, and the access cannot be fixed in a 3DS mode CFW like Luma3DS, right?

One (very, very dangerous) option that comes to my mind is replacing the NAND header with a fixed one once Luma3DS is running, and replace it back upon shutdown. But, better forget about that, that would in essence be a guaranteed brick given that a force shutdown or battery running out is a certainty over time.

AuroraWright, 2016-05-09:
Yeah the only way to fix is (possibly) would be what Tux said... if the partitions are encrypted with different key data, TWL FIRM needs to decrypt them to even work.

RShadowhand, 2016-05-09:
Moral of the story: Make sure to take backups and don't lose them.

TuxSH, 2016-05-09:
> we'd need a CFW running in TWL

Afaik there is no firmware on DSi/TWL, just a BIOS providing service calls.

llakssz, 2016-05-27:
Everyone who downgrades would have this problem? If not, what actually causes it?
Good that we can fix it with a backup though.

closed 2016-08-03 by TuxSH