LumaTeam/Luma3DS #87: Bootanim9 renders TWL_FIRM and AGB_FIRM unusable.

Boured, 2016-05-14:
As the title says above it make all the games that do load with them never show up. It doesn't even shoot out a system error, it just doesn't load.

Here is a video of it
https://youtu.be/dOWdgNLYn-w

I know it's this causing the problem because after uninstalling it, TWL and AGB worked fine and started with no issue.

TuxSH, 2016-05-14:
Is this related to [this BootAnim commit](https://github.com/Wolfvak/BootAnim9/commit/669bf93ab7128b9ab648250a3612adca978f48b8) and [the most recent one](https://github.com/Wolfvak/BootAnim9/commit/09add932b4c957dd9385066450a3fb8a290d0aba)?

Boured, 2016-05-14:
I honestly don't know much about commits, I just took the latest one (V 0.5) from the releases page.

TuxSH, 2016-05-14:
Oh. You need to use Luma's patchchanger then, or use the latest commit of BootAnim9 if it works.

https://github.com/AuroraWright/Luma3DS/wiki/Installation-and-Upgrade#optional-making-luma3ds-work-with-an-external-boot-manager

Boured, 2016-05-14:
Alright, I'll try it thanks ^^

Wolfvak, 2016-05-14:
This has already been fixed, and there's warnings everywhere saying that it won't work with Luma3DS unless you use the pathchanger (or one that includes patching).

I suggest either @TuxSH or @AuroraWright to close this

TuxSH, 2016-05-14:
No answer from OP, I'm closing

closed 2016-05-14 by TuxSH