LumaTeam/Luma3DS #88: C version of the pathchanger doesn't work correctly for paths shorter than "arm9loaderhax.bin"

Wolfvak, 2016-05-15:
The title is pretty explicit, but basically the issue lies here https://github.com/AuroraWright/Luma3DS/blob/master/pathchanger/pathchanger.c#L89

You should do something like 
'memset(found + 12 + sizeof(payloadname), 0, 74 - sizeof(payloadname)); // Zero out the rest of the payload name'

It isn't an urgent issue by any means, it just threw me off when I was trying to implement it for BA9, and given this is the "official" version, it'd be nice to contain a correct implementation. I've checked the Python version and it appears to be correct, but I prefer C for these kind of things.

Sorry for the lack of PR, I'm too lazy :P

TuxSH, 2016-05-15:
`payloadname` is set to zero at this line: https://github.com/AuroraWright/Luma3DS/blob/master/pathchanger/pathchanger.c#L80 .

Wolfvak, 2016-05-15:
Hmm... of course, your sizeof(payload_name) is always set to 38, so nvm.

The code is correct, I don't know what I was thinking.

closed 2016-05-15 by Wolfvak