LumaTeam/Luma3DS #90: Patch to prevent certain game's from disconnecting the internet

cheatfreak47, 2016-05-16:
Something I've noticed is that a handful of games force the Wifi off while they are running.

One such example of this is The Legend of Zelda: Ocarina of Time 3D

Was wondering if that could be patched out.
A practical use for a patch like this would be that it would negate the need for people using NTR to run this via the debugger: write(0x0105AE4, (0x70, 0x47), pid=0x1a)

TuxSH, 2016-05-16:
I'll take a look when I have time ;)

jaxrogers2, 2016-05-19:
This is a "feature" added in 11.0 to hinder downloading Homebrew Launcher payloads from the Internet with OoT3D and Cubic Ninja. Only these two games are affected, so if you need a temporary fix you can just restore a 10.7 backup or use emuNAND on <=10.7 to play the game with NTR streaming.

fincs, 2016-05-19:
No it isn't. OoT has blocked internet access since 3DS system version 1.0. 11.0 only revoked CN's permissions to use SOC:u and http:u (it does not disconnect the 3DS from the Internet).

TuxSH, 2016-05-19:
I assume it uses SOCU_CloseSockets, although I'm not so sure about that...

TuxSH, 2016-05-21:
@cheatfreak47 do you have another example of a game turning off Wifi?

cheatfreak47, 2016-05-21:
@TuxSH Sure do, Star Fox 64 3D and I've also heard Monster Hunter 3 Ultimate does as well, but I don't have that last game to check it personally.

usagirei, 2016-05-22:
Monster Hunter 3 Ultimate has never turned my wifi off as far as i can tell, it has multiplayer, so i don't think it'd do it.
Just popped my cart in for good measure and the Internet is still connected (Blue Indicator on homemenu, wifi indicator on unit lit)

Cu3PO42, 2016-06-07:
Pokemon X/Y/OR/AS disconnect from WiFi unless you explicitly connect to the internet in game as well.

Kaisogen, 2017-05-01:
Is this a precautionary measure by games? I think it might be part of the system, as it happens with certain system apps too. System Settings, in particular, does this. I wonder what causes this across many apps.

Bappsack1994, 2017-05-27:
all NFC Games still disable the wifi, if wifi turned on, then NFC will be disabled

Shadow322, 2017-05-27:
Wrong NFC is enabled when WiFi is on and off when WiFi is off

TuxSH, 2017-05-27:
The culprits are NDM:U EnterExclusiveState and NDM:U SuspendScheduler.

Now, for an efficient implementation => soon.

TuxSH, 2017-06-05:
(∩ ͡° ͜ʖ ͡°)⊃━☆ﾟ

closed 2017-06-05 by TuxSH

LordShaid, 2017-08-11:
So how would I find and edit NDM:U EnterExclusiveState and NDM:U SuspendScheduler? I'm looking into this to try to prevent Pokemon Moon from auto-disconnecting from the wifi for NTR 3.6 streaming. Sorry if this has been solved, I'm very new to homebrew 3ds stuff and don't see a solution (at least one that I understand). The write command performed in the debugger does not do anything to help me. I am running 11.5 by the way