LumaTeam/Luma3DS #91: Some games yield System Update message

ccfman2004, 2016-05-19:
When trying to redeem the Darkrai Pokémon code in Pokémon X, I get the "A system update is required" error even though my emuNand is up to date.  I even re-ran the update system option and it says I'm up to date so there may be something else that Nintendo is using for system version detection that isn't patched or something.

I'm on emuNand 11.0.0-33.

TuxSH, 2016-05-19:
Are you using the latest commit?
http://astronautlevel2.github.io/Luma3DS

ccfman2004, 2016-05-19:
I thought that 5.4 was the latest and wasn't aware that there was something newer or that site since I can't compile this stuff myself.

It does work now with the afc6f51 commit.

closed 2016-05-19 by RShadowhand

TuxSH, 2016-05-19:
It's indeed the latest _release_

reeved, 2016-05-21:
I tried to use the latest commit and I still can't connect to the internet using the Player Search System on either Pokemon X or Omega Ruby. I moved the /3ds/ files where they should be and also put the .dat file in the root folder. Was there a step to updating that I missed?

TuxSH, 2016-05-21:
What about the friend list?

What entrypoint are you using? AL9H, AL9H with a boot manager or menuhax?

reeved, 2016-05-21:
okay, I updated Luma using Star Updater and everything is fine now. on a sidenote, what did star updater do that I didnt when I manually tried to update Luma? I'm using A9LH.

urherenow, 2016-05-21:
How exactly did you try "to use the latest commit"? Look for my sig on gbatemp. The rxtools compiling guide applies *for the most part* to everything. Don't know how long ago you first cloned the repo, but perhaps you needed to update your submodules and/or libraries such as ctrulib.