LumaTeam/Luma3DS #92: Frozen Friends List and Miiverse

peter1357908, 2016-05-19:
Hello, AuroraWright!
I completed the guide by Plailect and got my system an arm9loaderhax CFW Sysnand! [https://github.com/Plailect/Guide/wiki](url)
**However, after the completion, I found myself unable to open Friends List and Miiverse..** I found some similar topics about how another user forgot to unlink NNID before the procedures and I didn't unlink my NNID either...
I posted the issue on Plailect's guide, but he seemed to be not sure about what happened and referred me to ask you about this problem.
[https://github.com/Plailect/Guide/issues/134](url)
So, is there any way to fix this little imperfection?

Thanks a lot for your investment and contribution to 3DSHACKS society!

CuriousTommy, 2016-05-19:
Well, there was an updated patch for the FRIENDS module https://github.com/AuroraWright/Luma3DS/commit/03dc4fc425dd3d26faa1d38436454760412f8a89, however, I am not sure if it is related to your issue.

linexis, 2016-05-19:
had the same issue. start mii-maker and then mii-lobby. that fixed it for me.

peter1357908, 2016-05-19:
Thanks for the comments! I got it "fixed" and here is my experience:
I am actually on the latest hourly build of "afc6f5,"(Plailect also suggested me to update to fix the issue) but the problem remained there.
I found some answers elsewhere saying exactly the same as to open Mii Maker once and everything solves itself, but it didn't work for me.. Mii Plaza(or Mii Lobby, right?) can work just fine, and so does eShop. It's just Miiverse and Friends List not opening there.
Some other answers include "opening the Friends List and letting it sit for around 20 minutes and it'll fix itself," and I put my last faith into that solution as I write this comment... and after 5 minutes I just went back and checked, **it worked!**
The thing is, it might not be an "issue" at all. Maybe Friends List just had to spend some more time initiating itself after the whole turning-Sysnand-into-CFW-Sysnand  process and Friends List, as well as Miiverse work as intended now. (Whew....)
It seemed I just had to have more patience :)

closed 2016-05-19 by RShadowhand