LumaTeam/Luma3DS #93: problems with loading luma

wolflmg, 2016-05-21:
Okay, here's the thing. I bought a 3ds that already has cakes and homebrew on it, along with several 3ds games already installed. I want to put luma on it and I'm pretty sure I have the files and folders all correct, but not positive. I'm able to get luma to load up in homebrew, but every time I go to load it keeps taking be to the luma configuration screen. Any help would be great.

urherenow, 2016-05-21:
This is not a support forum. Ask these things on gbatemp. Check the locking tab on your SD card and make sure it isn't locked or missing. And that there is free space.

wolflmg, 2016-05-21:
Thanks I go check these. As for space. There is a ton of space.

closed 2016-05-21 by RShadowhand