LumaTeam/Luma3DS #95: Problems with pathchanger

Ultracoolguy, 2016-05-23:
Today I was trying to use pathchanger for BootAnim9(just like the guide said), I dunno why but neither the Linux version(I have Ubuntu BTW)nor the Python version. The Linux version without sudo just say some error code with "Permission denied". With sudo it says "command ./pathchanger not found", and in Python it appears to work successfully but when trying to launch DsiWare, it just blackscreen, meaning it not worked.

BTW I have a N3DS A9LH Updated SysNand and Ubuntu 14.04.2 LTS.

hartmannaf, 2016-05-24:
Is python installed and is executing allowed by file permissions?

By the way, in the next BootAnim9 Release you will no longer need to use the pathchanger, you will only need to set a flag on your sd.
He pushed the source for this feature, so if you compile it by yourself you will be able to use it before the official release.

Ultracoolguy, 2016-05-24:
Python is installed, but it doesn't allow me to put executing permissions.

And I already noticed the commit. But how do I compile it? Just download source and compile it with make,right?

urherenow, 2016-05-24:
CLONE the source, including submodules. Look for me on gbatemp and follow my sig to compiling rxtools. Once you're set up, you should understand how to compile most 3ds stuff.

Ultracoolguy, 2016-05-24:
But I don't have Windows(I can't install neither Image Magic nor Visual C++ librarys)!

Ultracoolguy, 2016-05-24:
Oops! Looks like there is a way more easy(http://luma3ds.ericchu.net)! Will close issue.

closed 2016-05-24 by Ultracoolguy