LumaTeam/Luma3DS #96: Splash screen too fast

dbenfouzari, 2016-05-24:
Hi there,

I followed Plailect's tutorial and I now have Updated SysNand working with a9lh + Luma3DS.
I would like to customize my boot with a custom splash image. 
I successfully get my image showing, but that's way too fast ! I nearly can't see it.

Of course, I put my splash.bin in /luma/ folder, that's why i'm able to say it works, but that's really too fast (like 100ms showing).

Any way to make it way longer ?

Thanks in advance ! :)

hartmannaf, 2016-05-24:
It's not possible with luma, at least not without modifying the source code, but you could use a configurable boot loader like bootctr9.

RShadowhand, 2016-05-24:
Use the latest hourly, that added some splash delay I believe.

TuxSH, 2016-05-24:
It's consistenly 3 seconds with the latest commit.

closed 2016-05-24 by TuxSH