LumaTeam/Luma3DS #97: Online connectivity in certain titles error.

rhg, 2016-05-25:
I haven't tested this much, but on AS booted from an emunand on the latest firmware version and latest luma3ds trying to go online gives an error stating that 'a system update is required'. @dark_samus told me on IRC that this is due to the launcher and that there is a patch for this but the update broke it. I was able to get online via https://github.com/yellows8/ctr-httpwn but a fix would be nice (and necessary for games such as smash).

piratesephiroth, 2016-05-25:
just switch to a more recent build

rhg, 2016-05-25:
I'm running the latest release as of today, but glad to know it's fixed
already.

piratesephiroth, 2016-05-25:
It's broken on the latest official release.
You need to use a more recent nitghly build.

http://astronautlevel2.github.io/Luma3DS/

Dimensional, 2016-05-25:
I've attempted that, but the most recent build prevents me from loading any cia files nor booting into emunand.

rhg, 2016-05-25:
I'm half-tempted not to close this issue until a new release is cut with this fix, but on the other hand, I'm about to go build-hunting and if I find a working one then I might say which and close.

closed 2016-05-25 by rhg