LumaTeam/Luma3DS #100: Enabling Dev mode causes 022-2454 on NNID logging-in and some other e-shop related access

MelonGx, 2016-05-28:
As topic.
Seemed it had been existed for a long time since AuReiNAND era.

rboninsegna, 2016-05-28:
Yep, it's "normal", and it was kinda documented last time I checked (as "breaks eshop access" though, while it actually conflicts with **retail NNIDs and Amiibos**) :)

TuxSH, 2016-05-28:
It's because we set UNITINFO to 2, see [here](https://github.com/AuroraWright/Luma3DS/blob/developer/source/firm.c#L333).

urherenow, 2016-05-28:
@TuxSH, not to hijack the issue and turn it into a programming lesson, but I thought a number in brackets "[2]" had something to do with an array.

Just a bit confused...

TuxSH, 2016-05-28:
section[2] is the 3rd FIRM section, for NATIVE_FIRM it's the ARM9 one.

RShadowhand, 2016-05-28:
@TuxSH Is there a way to get the verbose debug info on error screen without causing NNID issues?

TuxSH, 2016-05-28:
[Yeah](https://github.com/AuroraWright/Luma3DS/pull/19/commits/6351e5b8c3f04b1d08bb5fb4099fee67a76198a6) but it was decided to modify UNITINFO after some debate because it's needed for other stuff.

fincs, 2016-05-28:
Why not patch NNID/eShop to think UNITINFO is retail?

rboninsegna, 2016-05-28:
(Kinda OT)

It's pretty funny that Big N put some verbose details in retail Wii U errors though!

RShadowhand, 2016-05-28:
@TuxSH Could you visit IRC for a moment?

TuxSH, 2016-05-28:
Fixed in https://github.com/AuroraWright/Luma3DS/commit/2424865fa109610b81cf9babe351addd72f86cd7, I guess.

sykhro, 2016-05-28:
UNITINFO is sent via post to whatever Nintendo server, i.e. when accessing the e-shop. you could just patch the request header

TuxSH, 2016-05-28:
I made it multi-choice (no feature, stuff + ErrDisp patch, stuff + UNITINFO patch), that's simpler.

closed 2016-05-28 by TuxSH