LumaTeam/Luma3DS #99: Can't boot Updated SysNAND O3DS since v5.3.1

m45t3r, 2016-05-28:
I have a O3DS with Updated SysNAND v11.0.0-33, A9LH and Luma3DS. Since version v5.3.1 my 3DS does not boot, with v5.3.1 I just got a black screen, while in v.5.5 I got stuck in boot splash.

I can boot in emuNAND (v9.2) and yeah, I used Gateway downgrader in the past. However this current version of my firmware is from a backup I made before the downgrade process, using Gateway software.

MelonGx, 2016-05-28:
1. Dump CTRNAND with D9.
2. Mount and defrag CTRNAND on PC.
3. Restore defragged CTRNAND.

m45t3r, 2016-05-28:
Did it, still not working.

m45t3r, 2016-06-02:
Ok, feeling dumb now. After looking at [FAQ](https://github.com/AuroraWright/Luma3DS/wiki/FAQ-and-Troubleshooting), got it working again.

Any reason that this change in 5.3.1 was necessary, though? Or is it mostly a code cleanup?

Verequies, 2016-06-02:
I have an Old 3DS XL (Toshiba NAND) which is working fine with Luma3DS v5.5 booted through Homebrew Launcher, and a Old 3DS Non-XL (Samsung NAND) booting Luma3DS using the same method. However the Old 3DS Non-XL won't boot v5.5, while it can boot v5.4 perfectly fine. Any reason why this might be?

Verequies, 2016-06-02:
The last commit that works is c3ad7ed.

timbiotic, 2016-06-26:
I am stuck, I have the issue where I cant run past 5.3.  But the FAQ file likes are riddles with viruses and I cannot get the files referenced. Is there a safe mirror without Trojans?

Ultracoolguy, 2016-07-07:
They doesn't have trojans. Maybe your antivirus haved a false positive?

TuxSH, 2016-07-07:
Avira Antivirus, I assume?

Ultracoolguy, 2016-07-07:
Or timbiotic clicked an ad...

v0idst4r, 2016-07-24:
Lol

pbanj, 2016-07-25:
That or he has something that's replacing links

closed 2016-07-25 by TuxSH