LumaTeam/Luma3DS #101: 3DS no longer boots.

chessj17, 2016-05-29:
I haven't updated since Luma3DS ver. 5.3. I put the new luma3DS ver. 5.5 on my 3DS and despite having all the required files my 3DS does not boot. The 3DS turns on for a second then immediately turns off. Sometimes when turning on is doesn't turn off but will remain on a black screen. HELP

DaGamer12345, 2016-05-29:
Have you ever downgraded using the Gateway launcher in the past?

chessj17, 2016-05-29:
No never touched the Downgrader

DaGamer12345, 2016-05-29:
When you updated, did you remove your configuration file? If you didn't, I would try that and see if it boots while holding down select.

chessj17, 2016-05-29:
Didn't do anything. Same as before

DaGamer12345, 2016-05-29:
Have you tried redownloading the update? Maybe you got a bad download.

chessj17, 2016-05-29:
Happened even before i updated. I assumed updating might fix it. If it help i haven't messed with it in about 2 weeks. No idea what happened to make it stop working

DaGamer12345, 2016-05-29:
It could be your SD card going bad, I would try a spare and if it boots with that then RIP your SD card.

chessj17, 2016-05-29:
Thats what I was thinking so I'm trying that right now.

chessj17, 2016-05-29:
Still have no luck.

DaGamer12345, 2016-05-29:
At this point, if 5.3 works, I'd just stick with it and report the issue in the GBATemp thread. Your A9LH install might be wonky or it could just be a one-off issue.

chessj17, 2016-05-29:
Nothing works anymore. Not even 5.3. I cant boot at all

DaGamer12345, 2016-05-29:
Hm, I was stuck in a similar situation after doing a system transfer. I would attempt to use another A9LH-supported CFW to see if it works. As a test, you could also try booting the Decrypt9 payload, just to make sure your A9LH wasn't broken at all.

chessj17, 2016-05-29:
I've also noticed that every second boot my 3DS does something, doesnt necessarily work but does something. When trying to boot Decrypt9 I get the logo on the bottom screen but the top bugs out

DaGamer12345, 2016-05-29:
if you can at least start A9LH payloads I would build A9LH again with your OTP and try reinstalling it.

chessj17, 2016-05-29:
How can i install it if I cant even get into my system

DaGamer12345, 2016-05-29:
You can use SafeA9LHInstaller in A9LH. Just replace the arm9loaderhax.bin with the installer one. Decrypt9 booted so you can at least get into something.

chessj17, 2016-05-29:
Well it looks like I may have some sort of awkward brick that A9LH can't save me from, I can't boot into any payloads and have them usable and The 3ds doesnt turn off with the SD card in or Out on boot

DaGamer12345, 2016-05-29:
Just to be sure, by booting into payloads you're actually replacing the arm9loaderhax.bin on the root, and not using Luma3DS to chainload them, right?

chessj17, 2016-05-29:
I've tried both

chessj17, 2016-05-29:
Apparently some other people have had an issue similar to this and its either some form of corruption, Dead sd slot or sd card

DaGamer12345, 2016-05-29:
I would try going to GBATemp and getting help from their knowledgeable people. You shouldn't have completely bricked at all, 2 weeks ago everything was fine. You'd definitely get better help there compared to on here. As a last ditch effort, just keep trying different SD cards until payloads work.

KingMarcin, 2016-05-29:
Do you happen to have a firmware.bin file in the luma folder on your SD card? Try deleting it.

chessj17, 2016-05-29:
Tried it. Still nothing

chessj17, 2016-05-29:
One person on GBAtemp thinks it could be catastrophic hardware failure and that im SOL. My last chance would be to hardmod it and try a downgrade to before A9LH

piratesephiroth, 2016-05-31:
A friend of mine has a similar problem. Luma 5.3 works fine but anything newer just makes the 3DS get stuck in a black screen

m45t3r, 2016-06-02:
@piratesephiroth I think this is similar to the issue I am having: https://github.com/AuroraWright/Luma3DS/issues/99

closed 2016-08-03 by TuxSH