LumaTeam/Luma3DS #102: Infinite 3DS logo upon booting into game following save manager usage.

ForcedToRock, 2016-05-29:
Whenever I'd run a save manager, JKSave Manager and SaveDataFiler for example, after exiting a Pokemon game (OR), sometimes booting into the game will freeze on the 3DS logo. It doesn't happen all the time, it takes a shut down and power on to exit out of that problem. I'm on the latest Luma commit, 11.0 sysnand. I've noticed this happening at least for a month, so it's been an issue during previous Luma updates. I'm not even sure if they occurred before installing A9LH

onmyouji, 2016-05-29:
Do you happen to run an FTP server? 

I also had the same problem of freezing 3DS logo. Initially I thought it was caused by JKSM, but then I found out that FTP-GMX was the culprit.

ForcedToRock, 2016-05-29:
I do run FTP-GMX, and I just so happened to have installed that about a month ago. It would be that I exit the game, use a save manager, run FTP, exit out, go back to the save manager, exit, boot into the game, sometimes it boots successfully, sometimes it doesn't. I hadn't considered the FTP server to be a problem.

RShadowhand, 2016-05-29:
Get rid of FTP-GMX and install this:

https://3ds.secretalgorithm.com/ftpd.cia

onmyouji, 2016-05-29:
Yeah, I found out about it from this [GBAtemp thread.](https://gbatemp.net/threads/release-ftp-gmx-customize-your-ftbrony.410506/page-4#post-6375666) (thanks to user ladygugu)

It has something to do with libsf2d library used by FTP-GMX, apparently it causes some problems. Besides freezing problem when starting a game, some users also have problem accessing Homebrew Launcher.

ladygugu tried to remove the libsf2d part and it solves the problem. He/she posted the CIA, you can download it from that thread. In the end I decided to go back using mtheall's ftpd, there're many new commits to master since its last release.

RShadowhand, 2016-06-05:
Irrelevant to Luma. Closing.

closed 2016-06-05 by RShadowhand