LumaTeam/Luma3DS #103: Any chance of having the 300 title limit removed/changed?

z235, 2016-05-30:
As all of you probably know, for some reasons Nintendo decided to limit the home menu to only ever show the icons of 300 installed game titles at once, even when using folders. If you have more than 300 titles installed, the oldest entries disappear until you remove enough titles to get below 301 again. This obviously means that the old entries are not actually being 'deleted', they are just hidden from the home menu. 

This looks* like something that maybe could be patched out (or at least the limit increased) by a CFW like Luma3DS. :)

I even found a [thread](https://gbatemp.net/threads/patching-the-300-title-limit-discussion-help-testing-needed.419029/) about this on GBATemp, handful of devs replied but it looks like it's been abandoned after derailing, for now.

It's probably a long shot but I hope any devs are interested in this and willing to take a look at it.

Please don't spam this issue with post like 'why' or 'nobody needs to have 300 titles installed' etc if you don't think its useful. 300 games really aren't all that much anymore, especially if you don't like juggling things around and have a large memory card and love creating custom VC games. ;)

I personally have about 140 3DS games installed, plus on average a 30 of each GB/GBC/GBA/SNES custom VCs, plus a few homebrew apps. That's right at the 300 mark, and I still have a few favourites I'd love to add to the collection.

Thanks for your time.

Zetchzie, 2016-05-31:
I think we all want this lol

OTFree, 2016-05-31:
Hopefully it's possible!

CoolFangs, 2016-06-18:
I recently upgraded to a 128gb microSD, unaware of this title limit. I was in for a rude awakening when I went over it. Hopefully something can be done about it eventually.

gudenau, 2016-08-16:
I am thinking that the limit should be kept on the 3DS, we could probably increase it on the n3DS because of the memory and CPU improvements. 300 was probably calculated and rounded down.

LiquidFenrir, 2016-08-16:
It's not a "title limit" but an **icon** limit. titles not showing can still be launched via FBI, DevMenu, GridLauncher and everything else that can launch titles.

Mrrraou, 2016-08-16:
I have a feeling that it would probably break all the *hax payloads

closed 2016-11-06 by TuxSH