LumaTeam/Luma3DS #104: Amiibo Reading Issue on o3ds

CrimsonMaple, 2016-05-30:
I have a pikachu amiibo and attempted to play with it on Smash and the amiibo was unable to be played. The console stated the amiibo was not meant for Smash. The amiibo had just been used on my friend's n3ds so it's not corrupted.

DonDalle, 2016-05-31:
The o3DS doesn't support amiibo.

Shadow322, 2016-05-31:
Yes it does with the 3DS NFC Reader/Wrighter for Nintendo 2DS, 3DS and 3DS XL :)

CrimsonMaple, 2016-05-31:
I have the amiibo reader :P 
Wouldn't have opened other wise.

TuxSH, 2016-06-03:
Did you try the 5.5 release build (not the dev one)?
It's likely an issue with UNITINFO set to the dev. value.

CrimsonMaple, 2016-06-04:
That was the issue, heh. Well then gonna close this! Thanks everyone taking the time to help my stupid mistake!

closed 2016-06-04 by CrimsonMaple