LumaTeam/Luma3DS #105: eShop update patch not working from 10.7?

vaguerant, 2016-05-31:
I'm not able to access the eShop directly using my a9lh'd USA New 3DS on 10.7 and Luma3DS 5.5. I have the updated sysNAND setting **un**checked, as I'm aware the patch is not applied when running current firmware based on this setting.

It's worth noting that the error does vary slightly based on whether this setting is checked, so the patch is doing something, just not quite enough to get me onto the eShop. With updated sysNAND **unset**, I get to the NNID login screen, then:
```
Error Code: 022-2511

In order to use online services, a
system update is required.

Go to System Settings to perform a
system update.

OK
```
With updated sysNAND **set** (i.e. patch not applied), I get the standard eShop update notice, before any NNID notice on-screen:
```
A system update is available!

To update, select System Update
in System Settings.

Come back after you've updated!

Cancel        System Update
```
It's worth noting that yellows8's ctr-httpwn continues to work perfectly well to allow me onto the eShop.

TuxSH, 2016-05-31:
You need to update NVER (I'm not doing this automatically since it'll break stuff like the FRIENDS patch used to).

vaguerant, 2016-05-31:
Ah, got it. Thanks for the clarification.

closed 2016-05-31 by vaguerant