LumaTeam/Luma3DS #106: [suggestion] allow use of out of region software keyboards

mistyhands, 2016-05-31:
I've spent quite a while trying to replace the system keyboard with the JP one, as it allows for both alphanumeric and jp input, to no avail. (changed title id, etc etc). As far as I understand, seeing as they have different titleids, different firmwares call for different title ids when the keyboard is needed. Hence, it should be theoretically possible to somehow modify this, so if we installed an out of region keyboard, the system would instead load that title, instead of the default.

fincs, 2016-05-31:
Should be a pretty simple NS sysmodule patch.

mistyhands, 2016-05-31:
@fincs oh, nice. I'd attempt a PR if I knew anything at all about this, but I don't :|

urherenow, 2016-05-31:
+1

Considering I live in Japan, but use a US, console, I would love this feature! It would be nice to be able to use a Japanese greeting for my streetpasses...

mistyhands, 2016-05-31:
@urherenow For now, I think you can export and modify the extdata on PC to change the greeting.

MaiconErick, 2017-09-17:
Besides in game custom built keyboards, I know that USA 3DS doesn't have support for Korean input.
It doesn't even display the title of the game if you make a homebrew with Korean characters. Just a bunch of black squares with "?". It has support for Japanese, though.

fincs, 2017-09-17:
Korean consoles have a special system font with Korean characters. None of the other versions of the system font have Korean characters.

TuxSH, 2020-04-15:
Closing, tracker cleanup

closed 2020-04-15 by TuxSH