LumaTeam/Luma3DS #107: This isn't loading

Wolfman666, 2016-05-31:
Hi there,

I have an Old 3DSXL with 10.7E firmware installed on it. I've never tried installing a custom firmware before and haven't tampered with the existing firmware on the console.

I downloaded the latest set of files for Luma3DS and followed the instructions to copy the various folders to my SD card and put them in the correct place (alongside the ones that are already there to enable me to use OoT3DHax).

I then put my SD card back into my 3DS and loaded up the homebrew launcher. I could see the Luma3DS option at the bottom of the menu, but all that happens when I click on it is that the bottom screen goes red for a couple of seconds, and then the console goes black and all I'm able to do is to switch it off.

Any idea what's going wrong?

jmarcelocjr, 2016-05-31:
You need to downgrade to 9.2 before booting up the luma

Wolfman666, 2016-05-31:
Ah yes, that might help haha. I probably should be trying this when I'm not so tired, but ah well. Thanks for that.

jmarcelocjr, 2016-05-31:
Don't forget to create the emunand after downgrading! :)

closed 2016-06-01 by TuxSH