LumaTeam/Luma3DS #108: Patch to AGB_FIRM for VC color correction?

JasonPK, 2016-06-01:
According to this thread on GBATemp, https://gbatemp.net/threads/restoring-original-colors-to-gba-vc.427642/, it's theoretically possible to patch the VC footer in AGB_FIRM so the color correction applies to all GBA games instead of having to hex edit each game individually. All of the GBA games look so dimmed and washed out, if the color correction could be done it would be awesome.

BlaX722, 2016-06-01:
This would be great! Add this if possible... please!

piratesephiroth, 2016-06-02:
Those are the regular colors, as if we were playing on an emulator.

It would look nothing like a GBA.

cheatfreak47, 2016-06-02:
@piratesephiroth Not actually true. 
AGS101 GBA SP or especially a DS Lite, or Gameboy Micro, the colors look a lot brighter, more close to the "saturated" look you're referring to.
I'm not saying there's anything wrong with it, I'm just saying it looks like washed out crap compared to the corrected colors. So I'm in favor of this as well. 

Would be awesome as an optional patch in the Luma 3DS config, just like the GBA Intro Patch.

JasonPK, 2016-06-02:
Let's just say the original GBA screen wasn't a pioneer in LCD technology, so just because the colors in AGB_FIRM right now emulate the "original experience" it doesn't change the fact that most of the games look dimmed and washed out on the 3DS screen. They could look way better.

vaguerant, 2016-06-02:
Yeah, there are any number of ways to play GBA games that aren't "on the original model GBA". The Game Boy Player, either SP model, the Micro, either DS or DS Lite; to say the dimmed VC graphics are most accurate is to ignore the vast majority of ways that it's possible to play GBA games which say otherwise.

joesteve1914, 2016-06-08:
This would be great, especially since a lot of people have lost their gba cia files and would need to make a whole new cia to fix the color.

Youngsie1997, 2016-06-08:
Correct me if i'm wrong but i believe there are a few games that the color fix causes issues with, i believe in the thread there's discussion of a game refusing to save, 

 If it's any conciliation  the process has been made easier via a customized footer generator that somebody has posted that also removes ghosting https://gbatemp.net/threads/restoring-original-colors-to-gba-vc.427642/page-5. (Of course you could make it a toggle option like the boot screen) but again this is assuming it's even possible to patch the firm.

joesteve1914, 2016-06-21:
@Youngsie1997 
That's why it should be an option in th configuration menu, like the n3ds clock speed and gba boot screen. 

Of course this might not even be possible. I guess we won't know unless it gets added to Luma3ds.

Youngsie1997, 2016-06-21:
I've had to remake all my vc anyway so I've just been using the footer generator works fine for me i'm not fussed about having it as a Luma option.

closed 2016-08-28 by TuxSH